opentx/opentx #7428: LUA model.getGlobalVariable() crashes when optional FM is not given

lshems, 2020-03-08:
bad argument #2 to 'getGlobalVariable' (number expected, got nil)

option 2 is flightmode, and is optional according to documentation.

Opentx 2.3.5

3djc, 2020-03-08:
Code exemple ?

kilrah, 2020-03-08:
Doc is incorrect, argument isn't optional

3djc, 2020-03-08:
We do realise that the published doc is largely outdated, so for now, best is to look at the source code

closed 2020-03-08 by 3djc

3djc, 2020-03-08:
Which just got fixed by Kilrah : https://opentx.gitbooks.io/opentx-2-3-lua-reference-guide/content/

kilrah, 2020-03-08:
Well no, the content is still unchanged

lshems, 2020-03-08:
OK,

It would be nice to get the doc up. Thanks.

Would you mind if it got set to optional?

In lua you just put a getFlightMode() on the second parameter, but it is not that nice.
Could try to make a pull request some time.

3djc, 2020-03-08:
No reason for that’ even more if you assume current is the default, where I would have assumed 0, so keeping it mandatory is probably sensible

lshems, 2020-03-10:
Ok, I'll not bother to try create a pull request to default the flightmode to the current flightmode, and keep the parameter optional according to original documentation.

getFlightMode(mode) works with an optional mode parameter, but the doc is not mentioning optional, where it is. Defaults to current flight mode (which makes perfect sense to me).

So I don't understand why you would ever think 0 is the default flightmode for the optional 'mode' in the getGlobalVariable??

ticket is closed, retorical question.

commit e26abb56 (2020-03-08): Fix #7428
commit 21a08edf (2020-03-08): Fix #7428

(cherry picked from commit e26abb5650ebf72fac672b04a2ea3d65a06fd0bb)