opentx/opentx #7430: Horus X12S Antenna warning after flashing to 2.3.5 and selecting Int. module

jclaver, 2020-03-08:
This is X12S with 'EU' enables in settings.
Just tried the above twice and got the same outcome (Antenna error haivng selected internal module) prior to trying to bind to an X8R Rx. I then installed OpenTX 2.2 and deprecated back to 2.2.4 with the matching SD contents. Tried binding again - I now get no antenna error but neither can I bind to the X8R.  Both of my X8Rs I've flashed to current but perhaps I have to go back with those too?  My X12S is an early model so not sure whether it can support e.g. FrSky Access in hardware terms.  I just donated £100.

jclaver, 2020-03-08:
Latest is I can't get the X12S to bind with anything - either ACCESS or D16 (GRX-8 or X8R).

3djc, 2020-03-08:
Have you replaced your internal module by an isrm upgrade in your x12s ? Because if not, your x12s cannot do ACCESS. And if you have, then you cannot use 2.2 anymore.

Please detail your x12s setup, and what firmware you flashed

jclaver, 2020-03-08:
Not that I know of no. UPDATE: My X10S can bind to both RXs. So why can't the X12S bind to an X8R anymore? ISRM upgrade?  No idea this existed so I will look into it thanks!

3djc, 2020-03-08:
If you have not upgraded your x12s and see access choice on your radio, it means you flashed a wrong firmware with internal access option that you should NOT select

jclaver, 2020-03-08:
No I didn't use ACCESS for binding the XR8 - I used D16

3djc, 2020-03-08:
If you have access choice it means wrong firmware. Please include a screenshot /photo of you radio/version screen

jclaver, 2020-03-08:
I'm on 2.3. (going back to 2.2.4 then to 2.3 fixed the antenna error)

jclaver, 2020-03-08:
Does the X12S use the same ISRM board as the X10?

3djc, 2020-03-08:
Please try to respond in details to our question, it will help us help you

jclaver, 2020-03-08:
[image]

3djc, 2020-03-08:
Given the Modules / RX version is present, I would think you have installed a firmware with internalaccess option selected. Could you flash without that option ?

jclaver, 2020-03-08:
I tried (2.2.4) it cured the antenna error issue but then I still couldn't bind to an X8R that works fine on the X10S Express so no - other/earlier firmware did not fix this

jclaver, 2020-03-08:
I'll upgrade to ISRM - Horus X12S support seems to be waning... pity cos its beautiful ergonomically

jclaver, 2020-03-08:
Or its possible I do have an antenna issue?

3djc, 2020-03-08:
Why don't you try to flash a correct 2.3.5 firmware without internallacess ? Because without the ISRM upgrade, it will create errors and not work correctly

jclaver, 2020-03-08:
Yes I could do that.  I've ordered the upgrade module though ;-)

jclaver, 2020-03-08:
Weird - I just did that and the bind did not work but I got audio back from the Tx saying 'telem recovered' but X8R Rx has red blinking LED not Green

jclaver, 2020-03-08:
So I created a new model and the binding worked OK!!!  - this is weird.  I think I'll upgrade to ACCESS

3djc, 2020-03-08:
OpenTX does not do anything binding related, we just tell the module ‘please bind’. Now it could not work with internalaccess option because the way to make that request is different between an xjt type module and an isrm. Please confirm that in version screen you are in 2.3.5 and the module /rx version line is gone. If that’s the case, you now have a correct firmware. Proceed with method and it should work. You can also get help from community forums like rcgroups or open RC forum. Closing this as it is clearly not an OpenTX issue.

closed 2020-03-08 by 3djc