opentx/opentx #7431: Center beep is played after boot even if input was already centered

nmaggioni, 2020-03-09:
**Describe the bug**

When **Center Beep** is enabled for an input, sound and haptic feedpack are played after boot has completed even if the said potentiometer was already centered.

**To Reproduce**

1. In the current model's setting turn on `Center Beep` for a potentiometer.
2. Ensure that `Pot positions` is set to have the chosen pot centered at boot.
3. Turn off the radio.
4. Center the pot.
5. Turn on the radio.
6. As soon as the splash screen goes away (and the model notes come up, if you have any), the radio briefly beeps and vibrates as the center beep does when crossing mid point.

**Expected behavior**

Center beep (and relative haptic feedback) should only play after the radio has booted and the mid point has been crossed once. Behavior should be similar to how the `!1x` frequency option works in Special Functions.

**General information**
 - OpenTX version: 2.3.5
 - Hardware: FrSky Horus X10S
 - ~~OpenTX Companion~~

bsongis, 2020-03-10:
@kilrah @3djc I am OK with this one, and the PR is good, I would like your opinion before pressing the button

kilrah, 2020-03-10:
I kind of consider current behavior expected as it gives you confirmation the control is properly centered and you're good to go...

nmaggioni, 2020-03-10:
@kilrah If you want a control to be centered before proceeding to me it makes a lot more sense to let the Pot Positions option warn you. The current behavior produces a redundant feedback that makes me wonder what could be wrong everytime I turn on my tx.

Moreover, the center beep will just not be played if the control isn't centered at boot (or at model selection time): that makes it a tad useless in the use case you've described.

PR #7432 (2020-03-27): Prevent center beep at boot
Fixes #7431, tested on my Horus X10S.
closed 2020-03-27 by bsongis