opentx/opentx #7436: New Timer triggers

SHezinger, 2020-03-11:
In my glider setup I use the throttle stick not only for the motor but also for the crow breaking.
At the stick positions from -10% to -100% the crow will be set (from 0% to 100% breaking) and 
at the stick positions from 10% to 100% the motor is running (from -100% +to 100%
Therefore I have to mixers with separate curves for crow breaking and motor speed.

The stick range between -10% and 10% has no function and is my default position (No crow breaking or motor). Unfortunately if I want to set a timer to Th% ,for example to log my motor runtime, the timer is always running with because the stick stands around 0%.

Possible Solution:
Allow to set the true output values CH1 to CH16 for the timer triggers.

Example:
If I want to trigger the Timer from my real motor output value (RETA setup), 
a trigger based on Channel 3 could be C3% , C3s or C3t

3djc, 2020-03-11:
Well you create a logicial switch that matches whatever yoou feel like, and use that as a trigger for timer ? Or do I miss something

SHezinger, 2020-03-11:
I could do that, but if use Th% and my stick is at 0% (motor off on my setup) the timer would be running with 50% speed because it thinks my motor is running with 50% throttle

kilrah, 2020-03-11:
Use CH3 as Throttle Source if that's your motor channel.

closed 2020-03-11 by kilrah

SHezinger, 2020-03-11:
Oh sorry, you are right! 
I missed that on my X12S. Maybe the name "Throttle" in the radio menu is a bit missleading for entries that are connectet to the timer

kilrah, 2020-03-11:
Well the Throttle source affects the Throttle timer which the THxx stand for...