opentx/opentx #7438: Jumper T16 auto off

ronbon172, 2020-03-13:
I recently left my radio on for a couple of hours and burnt out the internal module. Is there a way to auto shut off the radio or internal module to prevent burning it out accidentally?

3djc, 2020-03-13:
You have audio warning when not in use, and there is no reason that this would damage your module either

ronbon172, 2020-03-13:
Unfortunately I left the room and forgot about it until later that evening. after that I keep getting "emergency" mode when power cycling radio. Grayson Hobby confirmed with me that the module is burnt out.

3djc, 2020-03-13:
I have a lot of respect for Grayson Hobby, but a burned module would not trigger emergency unless something very very strange has happened. Have you tried a new sd card ?

ronbon172, 2020-03-13:
no

3djc, 2020-03-13:
That should be your first step, start with a blank, new, sd card, install sdcard pack (and only that), start the radio as new

closed 2020-12-10 by stale[bot]