opentx/opentx #7439: Not enough flash available on this board for all teh selected options

PedroCortinas, 2020-03-13:
Hi, Im havinga a prblem with the firmware download for a Taranis X7: (opentx-x7-lua-luac-es-2.2.4.bin) i hvae the "Not enough flash available on this board for all teh selected options" message...
can you help me?

Target0815, 2020-03-14:
Same vor Taranis X9D (without plus). Selected options: noheli, nogvar, lua. Also no luck without this options.

Imperativ, 2020-03-14:
same here.
Firmware on Taranis D9X: opentx-taranis-noheli-nogvars-lua-en-2.1.9
Companion 2.2.4 stops downloading after short time with the error message:
**Not enough flash available on this board for all the selected options**

Happens, no matter wich hardware or option combination is selected
[image]

3djc, 2020-03-14:
Yes @bsongis has been made aware of this issue, I'm sure he will fix it asap

3djc, 2020-03-14:
Could you try again ?

Target0815, 2020-03-14:
It's fixed, I guess.  Thank you.

closed 2020-03-14 by 3djc

PedroCortinas, 2020-03-14:
OK, perfect, thanks very much... all all right now...