opentx/opentx #7441: Horus X10S Express EU version Wrong PCB

Ghost-B3, 2020-03-15:
After OpenTX 2.3.5 flash into my new Horus X10S Express, I´ve got error message "Wrong PCB".

I´ve got EU LBT, mod 2 Horus X10S Express, so I´ve chosen "lua" and "eu", but it didn´t work either.
It uploads to the radio, but after turning it on, it says the error message "Wrong PCB".

I´ve searched for this issue, tried everything to make it work, but nothing helped.

Running 2.3.5 version of OpenTX.

3djc, 2020-03-15:
What is the exact name of the firmware you flashed ? (or what is the type of radio you selected in companion ?)

Ghost-B3, 2020-03-15:
I’ve selected FrSky Horus X10 Express / X10S Express.

>What is the exact name of the firmware you flashed ? (or what is the type of radio you selected in companion ?)

@3djc

3djc, 2020-03-15:
Thanks? We had a major issue that required us to qucikly change our firmware distribution servers. This cause a wrong 2.3.6 verison to be available for a couple of hours, and also probably the issue here. I'm trying to contact the person in charge of this. Will keep this thread updated

Ghost-B3, 2020-03-15:
>Thanks We had a major issue that required us to qucikly change our firmware distribution servers. This caused a wrong 2.3.6 version to be available for a couple of hours, and also probably the issue here. I'm trying to contact the person in charge of this. Will keep this thread updated

@3djc Is it repaired, or should we wait with flashing ?

3djc, 2020-03-15:
Repaired

3djc, 2020-03-15:
If you could also confirm the fix did work, it would be nice

Ghost-B3, 2020-03-15:
It's working.
Updated without any issue.

Thanks for fast answer and fix. :)

3djc, 2020-03-15:
Thanks for letting us know. Closing this then

closed 2020-03-15 by 3djc