opentx/opentx #7442: IRSM board not binding with ACCST receivers

DeeBee878, 2020-03-15:
I ugraded my Horus X10s RF board to the ISRM board.  I purchassed the upgrade kit from Horus RC who have been helpful and advised me to install back in November 2019 that I needed to instal OpenTx 2.3.3 followed by intalling the ISRM firmware found on the FrSky website under X10 Exspress.
It worked and I was able to bind ACCST and I test flew successfully in small foamie.  When I upgrades receivers to ACCST V2 as advised by FrSky those receivers would no longer bind to my X10s.
I upgraded 2 receivers to ACCESS firmware and they did bind and worked reliably.
I could not set the antenna to full time external however.  If I did, the external antenna would work but only until I switched  the Tx off.  Upon restart, it would give a defective antenna warning.

I have now installed OpenTx 2.3.6 and reinstalled ISRM V1.1.3 firmware as found on the FrSky website.  It binds well to my two ACCESS receivers but does not bind ACCST receivers either orignal V1 or the recent V2 firmware.
Range testing is good with extern antenna outperforming the internal by 3 to 4 RSSI at approx. 60m with TX  set on Range testing mode.
I no longer get a defective antenna warning although twice i noticed a message saying 'module upgrade required' has appeared when on an ACCST model setting.  Not sure if it will do that in ACCESS.
In summary, with 2.3.6, no ACCST D16 binding and ocassional message saying module upgrade required.  I am  not sure if it is an FrSky firmware issue or something that can be addressed in OpenTX.

kilrah, 2020-03-15:
1.1.3 for the upgrade ISRM is NOT SAFE and must not be used. That's why you get the "module upgrade required" message.

DeeBee878, 2020-03-15:
Thank you Kilrah for correcting me.
I did not realise what I had done.
I have installed ISRM 1.1.5 and I now have both internal and external antennas.
I can now bind ACCST V1 receivers successfully.
I still cannot bind ACCST receivers that have been updated to V2 ACCST.
Is this something only FrSky can address and is it safe to test fly on my current set up with OpenTx 2.3.6, or should I be waiting for further updates?
Dennis

3djc, 2020-03-15:
Yes, only FrSky can address binding accst v2 from ISRM

kilrah, 2020-03-15:
That is normal, V2 ACCST will be a new update from FrSky that's yet to come.

DeeBee878, 2020-03-16:
Thanks.  Done that and it seems to be working OK apart ACCST V2 of course

closed 2020-09-12 by kilrah