opentx/opentx #7451: input expo function set to diff when setting value to 0 in companion

lshems, 2020-03-20:
when setting the value of the expo to zero on any input, and saving the model, it will reset to diff as function type on next load.

**General information**
 - OpenTX version (e.g. 2.3.5)
 - Hardware (e.g. Frsky X9D+)
 - OpenTX Companion (e.g. 2.3.5)

kilrah, 2020-03-20:
That's intentional for eeprom savings.

lshems, 2020-03-20:
It is annoying though if you really want expo on an input, and use a script to set it on a different value than zero.

Now you have to set it to expo as well for the script to work.

Close it if you like, but it seems a drastic measure to save a bit.

kilrah, 2020-03-20:
This dates back from way before anyone would have dreamt of scripting, but is still valid...

lshems, 2020-03-20:
Closing is fine with me.

closed 2020-03-20 by lshems