opentx/opentx #7452: 2.3.5 failsafe not set after switch off and on the internal XJT module

janekx, 2020-03-23:
OpenTX 2.3.5 X9E 

Have switch ON the XJT module and set filesafe custom and set channels. 
Go to model and switch internal module XJT OFF.
Switch XJT internal module ON. 
FS is not set. 

Internal module FS setting is lost when switch OFF and ON module. 

As users wanted to switch off modules to prevent radiation doring long time model setup, is FS setting lost annoying when must set FS again.

3djc, 2020-03-23:
The same applies to all module data, not just failsafe (start, stop channel, receiver number,...)

kilrah, 2020-03-23:
To clarify the actual custom set values are kept since they are not in the module settings, but the global type selection is.

closed 2020-12-19 by stale[bot]