opentx/opentx #7455: Seperate rssi indicators for internal and external modules

joedunnaway, 2020-03-24:
When using the internal module as the primary and the R9M external module transmitting to a 900 mhz redundant receiver, RSSI values do not seem to corralate with anything when doing a range test.  It is unknown whether you are seeing values from the main receiver or the redundant receiver.
   RSSI Values for 2.4 ghz and 900 mhz would tell you the reliability of both links.

kilrah, 2020-03-24:
What radio? Most do not have separate telemetry lines that would allow this. External module needs to have telemetry set to disabled if the internal one is active.

joedunnaway, 2020-03-24:
Radio is the Horus x10s express.  Receivers are the rx8r with Accst and the r9mm with Access.  R9mm was set to "no telemetry".  There is no option to disable telemetry on the external module.  An rssi value is displayed no matter which receiver is active. Joe DunnawaySent from my T-Mobile 4G LTE Device

  external modules (#7455)

closed 2020-12-19 by stale[bot]