opentx/opentx #7456: Global function don’t work when programmed on the radio

d3lpy, 2020-03-24:
** Describe the bug **

I use a global function to adjust the speaker volume so that messages sent from the radio do not disturb my surroundings. This is a function that I have always used on all my radios for a very long time.
To know
FG1 ON Volume S1 [v] <- box checked
I also> checked global function [v] <at the model (s) level
This function works on my X9D + and the X9D + 2019 in OpenTX v.2.3.5
I have now passed my Horus X10S Express from FrOS 1.3 then updated to FrOS 2.x and finally to OpenTX 2.3.5.
By programming the global function from the X10Sxpress in the same way as I always did, it is "displayed" correctly on the radio, but does not work.
If I "transfer" the radio model from Compagnon, the global functions are "empty" !!!
Now if I program the function from companion and transfer it to the radio, the function works ???

  Reproduce
  By programming the global function from the X10Sxpress in the same way as I always did, it is "displayed" correctly on the radio, but does not work.
If I "transfer" the radio model from Compagnon, the global functions are "empty" !!!
Now if I program the function from companion and transfer it to the radio, the function works ???

  Expected behaviour
  Program my global functions from the radio.

*** Update LOG ***

C: / Program Files (x86) / OpenTX / Companion 2.3 / dfu-util.exe -a 0 --dfuse-address 0x08000000 --device 0483: df11 -DC: / Users / b3rna / OneDrive / Documents / OpenTX / HorusX10Xpress / opentx-x10express-lua-noheli-ppmus-fr-2.3.5.bin
=================================================== ================================
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Filter on vendor = 0x0483 product = 0xdf11
Opening DFU capable USB device ... ID 0483: df11
Run-time device DFU version 011a
Found DFU: [0483: df11] devnum = 0, cfg = 1, intf = 0, alt = 0, name = "@ Internal Flash / 0x08000000 / 04 * 016Kg, 01 * 064Kg, 07 * 128Kg, 04 * 016Kg, 01 * 064Kg, 07 * 128Kg "
Claiming USB DFU Interface ...
Setting Alternate Setting # 0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash"
Downloading to address = 0x08000000, size = 1064580
.................................................. .............
File downloaded successfully
No valid DFU suffix signature
Warning: File has no DFU suffix

=================================================== ================================
On Windows 10.

bsongis, 2020-03-24:
Checked here on XLiteS and X10 Express. No issue.
Please check the nightly builds.
If your issue is still there, please share your general / model settings so that we could check what's wrong

d3lpy, 2020-03-24:
désolé, j'écris en français, je ne maîtrise pas bien l'anglais.
je ne sais pas si tu as vu à la fin du LOG de la mise à jour l'indication suivante : 
_File downloaded successfully
No valid DFU suffix signature
Warning: File has no DFU suffix__
La mise à jour de FrOS que j'ai fait avant la MAJ optx2.3.5 est une béta !?
J'ai aussi refait une MAJ direct depuis la radio...
**sinon, je peux trouver où les "nightly" ?**

3djc, 2020-03-24:
English only please. And I told you already on the french forum that this message is not an issue

d3lpy, 2020-03-24:
I do not know if you saw at the end of the LOG the update of the following indication:
  File successfully downloaded
  No valid DFU suffix signature
  Warning: the file does not have a DFU_ suffix
  The FrOS update I did before the optx2.3.5 update is a beta

d3lpy, 2020-03-24:
Where I can find nightly after version 2.3.5

3djc, 2020-03-24:
Assuming you sue windows : https://downloads.open-tx.org/2.3/nightlies/companion/windows/

d3lpy, 2020-03-24:
Yes but this is for companion.... i don't have a problem with companion but with OPTX2.3.5 ?!
If i use compagnion 2.3.6, when i can download the firmware, is 2.3.5 ??? it's realy this i must test ?

3djc, 2020-03-24:
Download companion nightly, set you release channel to nightlies, download a firmware and flash it. You can seek help for that in the french forum

d3lpy, 2020-03-24:
Do you have the link for French forum ?

mpjf01, 2020-03-24:
Getting back to the volume control issue. I have tested my X9D, X10S, X12S all with the same settings and model. The global function for volume control works as intended on the X9D and X12S. It does not function on the X10S, the volume is always at approx mid level.

3djc, 2020-03-25:
https://photos.app.goo.gl/Tj8SiGcbtj7Agiqr8

3djc, 2020-03-25:
Closing, was user error ;)

closed 2020-03-25 by 3djc