opentx/opentx #7459: Jumper T16 USB stopped working with OpenTX 2.3.5

MrPipa, 2020-03-25:
**Describe the bug**
When connecting the Jumper T16 via USB, Windows reports "device descriptor request failed" or nothing at all.
Things I've tried to fix the issue:  
 - Reinstalling the driver in Zadig
   - leads to an error that the driver failed to install
 - Re-flashing the firmware (OpenTX 2.3.5) from the SD Card directly, without Companion (since that is no longer an option)
 - Formatting the SD card and putting fresh, non edited content on the SD card
 - Connecting the radio while it is off
 - Connecting the radio while powered on
   - with the USB Joystick option
   - with the SD Card option
   - with the Serial option (set in system settings)
 - Tried a few other Windows machines

**To Reproduce**
1. Flash OpenTX 2.3.5 to the Jumper T16 (no internal multiprotocol module, non-pro version)
1. Put the 2.3.5 SD Card contents on the SD Card and insert into the Jumper T16
1. Disconnect USB
1. Turn on the Jumper T16 and check that it's booting/working.
1. Connect USB, select SD Card mode on the Jumper T16

**Expected behavior**
The computer connects successfully to the Jumper T16 and companion is able to read the firmware and/or SD card content.

**Actual behavior**
The computer does not see anything connected or reports "device descriptor request failed".  Companion is not able to read the firmware nor the sd card content.

**General information**
 - OpenTX version: 2.3.5
 - Hardware: Jumper T16
   - internal multiprotocol module: No
   - external multiprotocol module: Yes
   - non-pro version, this is the plain stock (first) model
 - OpenTX Companion 2.3.5 on Windows 10

kilrah, 2020-03-25:
Hardware issue, check ribbon cables.

closed 2020-03-25 by kilrah

MrPipa, 2020-03-25:
Thank you very much 👍
The USB is working again after re-plugging all the ribbon cables from the USB port to the main board.

run-towards, 2020-03-27:
Had the same issue today. There also was 'Key stuck' error message. 
Resolved by flipping ribbon cable as @MrPipa described.

fabi9163, 2022-05-17:
Same problem. No USB device detected/Windows can't detect the device. Open the remote and jiggle the ribbon cable has solved the problem. (I already have installed the upgraded ribbon cables...sad...very sad