opentx/opentx #7461: No Subtype menu for MULTI / FLYSKY AFHDS2A

roemera, 2020-03-27:
**Describe the bug**
Hi, I just updated my jumper T12 to v2.3.5. Seems I can no longer switch my Subtype to PPM when using the MULTI / FLYSKY AFHDS2A protocol. Looks fine on the simulator, but on the radio there is no Subtype menu at all. When scrolling through the protocol menu, sometimes the Subtype menu will show up for a split second on FLYSKY AFHDS2A then disappear.

Thanks for everyone's hard work on OpenTX! I'm will to test any fixes for any bugs for the T12.

**Expected behavior**
Subtype menu is shown.

**Screenshots**
[image]

**General information**
 - OpenTX v2.3.5
 - Hardware Jumper T12
 - JP4IN1 v1.3.0.76

roemera, 2020-03-27:
So, same issue on the nightly build, tried 2.3.6N165 (4e5cd96a)
With both versions if I am fast enough when toggling the Mode, I can quickly switch the Subtype. I was able to get my TGY-iA6B outputting PPM. Took about 10 tires to get the menu switched quick enough.
[image]

3djc, 2020-03-27:
Your module shows version 1.2.1.84, not 1.3.0.76

roemera, 2020-03-27:
Wow, can't believe I missed that. I did try to flash 1.3.0.76 today. Let me try again in the morning, thanks.

roemera, 2020-03-27:
@3djc got my multi flashed to 1.3.0.76 and everything seems to be working now!
Is there somewhere that lists which multi version(s) each OpenTX version works with?

3djc, 2020-03-27:
OpenTX asks you to update the module when it is needed :)

closed 2020-03-27 by 3djc