opentx/opentx #7467: (2.3.6) Emergency mode when entering themes

d4rkf0x, 2020-03-27:
OpenTX enters emergency mode when navigating to system settings -> SD card when the background.png image is changed. However, the background.png replacement is pulled up normally on the main menu.

Steps to reproduce the behavior:
1. Change the background.png in SD/THEMES/Default to a different background image.
2. Power off T16, disconnect from the computer (if you do not power off and disconnect, then the issue will not occur).
3. Turn on T16, the new background image is displayed, as expected. No issues except when doing one thing.
4. Hold the SYS button until the system settings menu appears.
5. Press the Page button
6. Emergency mode activated

**Expected behavior**
T16 should show SD card contents as normal. If the default background image shouldn't be changed, then a warning at startup to let us know that this isn't a bug would be expected as well.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version: 2.3.6
 - Hardware: Jumper T16 Pro V2
 - OpenTX Companion: 2.3.6

kilrah, 2020-03-28:
Can you post your background.png file (zipped so it doesn't get modified)?

kilrah, 2020-03-28:
Fixed in https://github.com/opentx/opentx/commit/771c6ea3bdd197d4943f6470cb9d57af30ba995b

closed 2020-03-28 by kilrah