opentx/opentx #7475: Create Windows start menu shortcuts for all users instead of "current" user

pre-martin, 2020-03-31:
When installing the Companion under Windows, the installer requires admin privileges. During the installation process it creates start menu entries for the "current" user, which is the Administrator account, if your own user does not have admin privileges (which it never should have).

In consequence, that start menu entries are only visible for the Administrator but not for all other users of the system.

**Expected behavior**
Start menu entries are created for "all" users so that each user is able to start the Companion through the start menu.

**General information**
 - OpenTX Companion up to and including 2.3.7 on Windows

**Note**
I am able to create a pull request for the installer. The PR would modify the creation of the start menu entries and it would be backward compatible in the uninstall process, so that start menu entries in the old and new location will be removed.

3djc, 2020-03-31:
At best it should be a choice, clearly not mandated for all users

kilrah, 2020-03-31:
> if your own user does not have admin privileges (which it never should have)

That's very debatable, since UAC was introduced there is no reason to run on a standard account anymore.

pre-martin, 2020-03-31:
I will add a page in the installer with radio buttons: Create start menu entries for (a) current user (b) all users, with "current user" preselected (= the current behaviour).

Would this be okay for you?

3djc, 2020-03-31:
Sounds good to me

kilrah, 2020-03-31:
Yes fine

closed 2020-04-09 by 3djc