opentx/opentx #7476: Can't download firmware

flabrosse, 2020-03-31:
I'm new to OpenTX.  I recently received a Jumper T16 and have started using companion.  Version 2.3.5 and as of yesterday 2.3.7.

Every time I start Companion it wants me to download the firmware.  I say yes and this results in "Unknown server failure, try later".

The latest I tried was opentx-t16-&internalmulti-&lua-en but I have tried a variety of options, since 2.3.5, and now 2.3.7.

Pressing on the Release Notes button leads to a 404 error.

I have googled and searched through the issues here for the error message and only found one mention to old firmware versions, nothing to the current ones.

kilrah, 2020-03-31:
Check that companion is set to the release channel.
Also is that the exact finlename with the "&"s in the name? This is incorrect, should be "opentx-t16-internalmulti-lua-en"

flabrosse, 2020-03-31:
Thanks for the quick answer!

This is the name given in the window.  The URL I get in the console is [https://downloads.open-tx.org/2.3/release/firmware/getfw.php?fw=opentx-t16-%26internalmulti-%26lua-en.bin](https://downloads.open-tx.org/2.3/release/firmware/getfw.php?fw=opentx-t16-%26internalmulti-%26lua-en.bin). If I go to that URL I get an error 127.  If I remove the '&' (%26), then it does download something.  Using wget removing the %26 also gets me something that could be the firmware (binary file). 

Companion is set to the release channel.

3djc, 2020-03-31:
What OS are you using ?

flabrosse, 2020-03-31:
Ubuntu 18.04.

The name it puts in the save box is `opentx-t16-&internalmulti-&lua-en-2.3.7.bin`.

3djc, 2020-03-31:
Are you compiling yourself ?

flabrosse, 2020-03-31:
No, downloaded from the OpenTX [site](http://downloads.open-tx.org/2.3/release/companion/linux/companion23_2.3.7_amd64.deb).

3djc, 2020-03-31:
Just tested debian, looks fine, installing ubuntu

kilrah, 2020-03-31:
Try deleting the radio profile and creating another

flabrosse, 2020-03-31:
In fact, sorry, kubuntu 18.04.  Don't think it makes any difference though.

flabrosse, 2020-03-31:
Just created a new profile (without deleting the previous one, does it make a difference?) and the same happens.

flabrosse, 2020-03-31:
If I unselect all build options (which by the way happens every time I get into settings, I need to report that), then the '&' are not added.  It's only when build options are selected that the '&' are added.

3djc, 2020-03-31:
Just installed fresh Ubuntu 18.04, works fine

flabrosse, 2020-03-31:
Even when selecting build options?

3djc, 2020-03-31:
[image]

flabrosse, 2020-03-31:
With the same options:

[image]

3djc, 2020-03-31:
There must be something wrong somewhere in your setup, working fine here right after default install

[image]

flabrosse, 2020-03-31:
Given that it works without build options but not with build options, that points to a bug in the software, probably relating to my setup.  Closing seems a bit excessive.  Believe me I appreciate what you do for the community!  I do write software so will have a look when I have time, but please don't close it.  Others will have the same issue.

kilrah, 2020-03-31:
Works fine here with ubuntu 19.10 too, and we never had a similar report even though this is years old. I looked at the code and don't see where something could be wrong, so it's likely some pretty unique interaction with QT and whatever you're using as desktop enviromnent. 
It's unlikely it's something we can investigate...

flabrosse, 2020-03-31:
I'm downloading it and will compile from sources to see if this still happens.  If it does, then I'll have a look.  This will obviously take time as I don't have much time (even these days!) and I don't know the code at all.

flabrosse, 2020-03-31:
"Good" news.  The self compiled version has the same behaviour as the pre-compiled one.  I'll work on it, but it will take time.

flabrosse, 2020-03-31:
Quick investigation revealed that if I manually remove the '&' from the paths in the config file then all is fine.

If I create a new radio profile then the '&' are there.

Can someone reopen the issue please and I'll find a solution.

bsongis, 2020-03-31:
@flabrosse thanks for your proposal!

flabrosse, 2020-03-31:
I found the problem (but not the solution yet).  The name is made by concatenating a base name with a language and build option names, all separated by dashes '-'.  The build option names are obtained by getting the names of the corresponding QCheckBoxes.  The name contains an '&' as a shortcut, which is returned by the function text().

I found a couple of references to that, using PyQT, but apparently C++ suffers in the same way.  This seems to be due to a recent QT update (maybe a bug).  This is not documented in the QT docs.

I can see two solutions and would love to have some input from developers of the project:

Option 1: just parse the string and remove the '&'.  Simple and local, no global change required.

Option 2: not use the names of the QCheckBoxes.  Advantage is that they could use a more explanatory description.  The downside is that a lot of code (I guess) will need to be modified where the QCheckBoxes are created and used.

What do people think?

flabrosse, 2020-03-31:
Fixing as Option 1 also fixes the bug where the build options appeared not be saved, obviously.

Because of that extra bit that I had not thought about, Option 2 is even less appealing.

Suggestions?

kilrah, 2020-03-31:
The question is why does the name return an & on some systems and not others when it's the same code?

flabrosse, 2020-04-01:
Good question.  The one discussion I saw online mentions a recent update of QT.

The most relevant package I think is:

```
Package: libqt5gui5
Version: 5.9.5+dfsg-0ubuntu2.5
```

Is that what others have?

flabrosse, 2020-04-01:
I am ready to make a pull request with my option 1.  If people are happy for me to do it I'll do it.

kilrah, 2020-04-01:
Do you have links to that discussion? It looks like a QT bug then, and it's rather there that it should need fixing.

flabrosse, 2020-04-01:
Apparently that appeared in QT 5.4: [https://riverbankcomputing.com/pipermail/pyqt/2015-October/036499.html](https://riverbankcomputing.com/pipermail/pyqt/2015-October/036499.html)

My Ubuntu is up to date LTS, so not supporting that up to date version ot QT could limit a lot.  Also, the fix is trivial so not having it makes not much sense, IMHO.

kilrah, 2020-04-01:
Can't just be that then, would have been an issue for years. 
Back to something weird with your environment. What exactly are you using as desktop environment and theme? In the past there was already an issue once where it was the user's UI theme that was broken.

flabrosse, 2020-04-01:
I'm also not convinced it's a bug.  If you say button.setText("&lua"), you would expect that button.text() returns "&lua".

My Option 2 has the advantage that the software would be closer to an MVC model, while at the moment the UI is the data, with very tight integration.  But it would be a big change and probably way beyond that simple bug that consists of adding `.remove(QChar('&'))` at the end of a line.

kilrah, 2020-04-01:
> I'm also not convinced it's a bug. If you say button.setText("&lua"), you would expect that button.text() returns "&lua".

Except that we don't put any &s in there.

flabrosse, 2020-04-01:
I'm on kubuntu, with a standard KDE theme (Breeze Dark).

flabrosse, 2020-04-01:
> > I'm also not convinced it's a bug. If you say button.setText("&lua"), you would expect that button.text() returns "&lua".
> 
> Except that we don't put any &s in there.

Interesting!  My companion definitely has shortcuts on almost everything on the Settings dialog.

I read somewhere that shortcuts could be automatically created to the first letter if not already set as a shortcut.  Some on the dialog are not the first letter.  I can't find that at the moment.  I'll investigate further.

flabrosse, 2020-04-01:
It's a feature!  And other [projects](https://github.com/GoldenCheetah/GoldenCheetah/wiki/Working-around-KDE-shortcut-injection) have the same problem.

Given that it's a feature of some environments and that the fix is easy, should we just do it?

3djc, 2020-04-01:
Then fix should be on your environment:

add

[Development]
AutoCheckAccelerators=false

to ~/.config/kdeglobals

flabrosse, 2020-04-01:
But it's not a bug in the environment, it's a feature!  Why not supporting it when it's so trivial to do it?  Other people will have the same problem (I'm not the only KDE user around!).

PS I don't use the feature, I never noticed it.

flabrosse, 2020-04-01:
A lot of people are complaining about that feature (eg [link](https://bugs.kde.org/show_bug.cgi?id=337491) and a fix was introduced in 2019 for applications that don't explicitly link with KXmlGui.  Presumably companion uses xml gui.

I'm happy either way, but if nothing is done to make the code work, this should be documented somewhere, IMHO.

kilrah, 2020-04-01:
So it's a bug in KDE that was fixed more than one year ago... no reason we'd have to add special handling for that.
This issue will be there to point people to should someone else have the problem.

closed 2020-04-01 by kilrah