opentx/opentx #7479: Crossfire RSSI warning improvements

JBKingdon, 2020-03-31:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
The warning and critical alerts for RSSI on crossfire are hard-coded to the link quality sensor, and the maximum thresholds are limited to 75 and 72 respectively. I would like to be able to use the rssi sensor for warnings, and if using LQ, then I would also like to be able to set a much higher warning threshold.

**Describe the solution you'd like**
The UI for rssi warnings has a selection for the sensor to use but for crossfire it is fixed to "Default". It would be great if we could choose between rssi and LQ here. If using LQ, the midpoints used to bias the 6 bit warning and critical fields could usefully be raised from 42/45 to maybe 68/69 since LQ can fall very rapidly once it starts to drop.

**Describe alternatives you've considered**
Keep it as is, people don't seem to be complaining  :)

**Additional context**
I was investigating this in the context of ExpressLRS which uses the crossfire protocol, trying to figure out what the appropriate range of values to send for RSSI was in order to work well with the existing rssi warnings. However I couldn't get warnings to work with RSSI at all.

kilrah, 2020-04-01:
You can disable the built.in alarms and use logical switches/special functions to do whatever you want on any sensor

JBKingdon, 2020-04-01:
> You can disable the built-in alarms and use logical switches/special functions to do whatever you want based on any sensor

Yes, absolutely. I set that up last night and it works, but it doesn't seem possible to completely replicate the built-in warnings, the SF for playing sounds appears to only have access to a subset of the available sounds. Perhaps going to lua scripts would allow a more complete emulation. But since we have the functionality built-in, doesn't it make sense to improve that rather than try and emulate it another way?

3djc, 2020-04-01:
You can play any sound you want. If the one you want are not available in the list, just add them to your sound folder

JBKingdon, 2020-04-01:
> You can play any sound you want. If the one you want are not available in the list, just add them to your sound folder

That's good to know. So the existing warning sounds are in SOUNDS/en/SYSTEM - I should copy the ones I want to use to the top level sound folder or somewhere else?

3djc, 2020-04-01:
Copy (don’t move) the one(s) you want from SOUNDS/en/SYSTEM to SOUNDS/en

JBKingdon, 2020-11-26:
Just noting that as the ExpressLRS community grows we're seeing an increasing number of questions around why the rssi warnings behave the way they do. Being able to select between rssi and lq would be really nice to have.

MastroDado, 2021-02-13:
I fully agree, would be nice to be able to switch rssi or lq for the default warning. And to be able to increase the value in case of lq monitoring.

closed 2022-04-17 by stale[bot]