opentx/opentx #7488: Frsky x-lite trainer option missing

matthewthorp, 2020-04-02:
Trainer option in the model menu missing from opentx. I've recently bought a Bluetooth module to add to it as in this video.
https://www.youtube.com/watch?v=fyT6XohZIsY&t=478s

Gone through version history and seems to be showing last in v2.2.2

General information
 - OpenTX version 2.3.7
 - Hardware Frsky x-lite
 - OpenTX Companion 2.3.7 on windows

kilrah, 2020-04-02:
Did you add your module yet? If the module isn't detected it won't show up.

matthewthorp, 2020-04-02:
Haven't yet, but it won't even show up in companion.

3djc, 2020-04-02:
No it does not show in companion because most x-lite do not have this mod, and if we where showing it, then people would either complain or assume their radio is supposed to have it.

So best course of action is installing it in your radio, it should get detected and relevant menu appear on your radio (and no companion will still not show it for the reason above)

closed 2020-04-02 by kilrah