opentx/opentx #7490: no checkmark for Disable Telemetry on x9 lite italian on RX OPTION

alexdibi, 2020-04-02:
no checkmark for Disable Telemetry on x9 lite italian on RX OPTION

**To Reproduce**
Steps to reproduce the behavior:
1. Go to model page
2. Click on receiver and select RX option
3. See error on disable Telemetry option

[image]

**General information**
 - OpenTX version: 2.3.7
 - Hardware: Frsky X9 Lite
 - OpenTX Companion (e.g. 2.3.7 on Windows 10)

kilrah, 2020-04-02:
Can you provide what you think is a good label that is short enough to fit?

alexdibi, 2020-04-02:
Telem. Off

kilrah, 2020-04-02:
And current one is good for wider displays?

alexdibi, 2020-04-02:
I have not tried

kilrah, 2020-04-02:
No I mean is it good to keep that sentence, is the wording good?

alexdibi, 2020-04-02:
yes, it can be fine, otherwise "Telemetria OFF" it should also be good for small displays and it would be good for both

closed 2020-04-18 by 3djc

robustini, 2020-04-20:
My fault, thanks for the fix.

robustini, 2020-04-20:
@3djc , check the fix, there's a doble "" before Telem.

[image]

3djc, 2020-04-20:
fixed

commit 8da4cb25 (2020-04-18): Fix #7490