opentx/opentx #7493: Misbehavior of the 6 pos switch on T16

giacomo892, 2020-04-03:
After some minutes of uptime, the model which was set in mode 6 switched back by his own to mode 5, as I would have pressed the button 5. I did not. So I started to investigate.

Basically the mode selection button number 6 starts to output the button 5 pwm value.
If button 5 is pressed afterwards it does nothing, just skipped.

If you continue to push other buttons they works (1 2 3 4). 5 does nothing and 6 outputs button 5 value.

Expected behaviour:
https://mega.nz/file/VpJR0IQR#QM6GQ2VpvZCReEuGJTk8C-_SDpKmtUvAQNRgfBSa8Xo 
What happens:
https://mega.nz/file/EtIzxQLT#KwGQ3KFmyKJ1WGgfya1mim2eA5-Ic6xzBasPVgHBCQA

I was able to reproduce this behaviour two times in a row but I cannot anymore.

 - OpenTX version 2.3.8
 - Hardware Jumper T16
 - Yaapu Telemetry 1.8.1 
 -  Using external module with PXX and getting telemetry back via Passthrough,

3djc, 2020-04-03:
Since the 6POS is not managed by OpenTX, it is either an ribbon cable issue, or another hardware issue

Xillius178, 2020-04-04:
I think that's an calibration issue from the T16... 

Please Restart Calibration... 
If you must turn the sticks in the calibration, press also all 6 position buttons...

You can see the Status of it in the middle (e.g. 1/2/3/4/5/6)
 
Please test it...

myozone, 2020-04-04:
Dirty contacts on the switch ?, I assume it's a resistor ladder to a analogue input ?

giacomo892, 2020-04-04:
@3djc I see your point here, are those managed by a little MCU with send an analog value to opentx?
@Xillius178 It's calibrated, since it works and then randomly start to misbehave
@myozone The 6 bottoms are on a big PCB, to contacts to be cleaned rather than a big strip which I've cleaned.

3djc, 2020-04-04:
Yes exactly. That little mcu manages the led and sends a 6 step analog value based on the last detected push

3djc, 2020-04-05:
Closing this as there is all indication that this is an hardwrae issue, unrelated to OpenTX

closed 2020-04-05 by 3djc