opentx/opentx #7494: OTX Companion Jumper T16 6P switch not changing icon

myozone, 2020-04-04:
I just discovered in Companion version 2.3.7 the 6 mode switch (6P) doesn't change the icon on the simulator screen - it stays at 1. This could be me not doing the right thing of course.

3djc, 2020-04-05:
Did you calibrate it ?

myozone, 2020-04-05:
This is entirely in OpenTX companion (I'm not sure you can calibrate it within companion), I don't even own a T16 yet. I was just getting used to it first before I eventually get one.

3djc, 2020-04-05:
You can calibrate without companion simu as you would on a regular radio

closed 2020-04-05 by 3djc

myozone, 2020-04-05:
> You can calibrate without companion

Eah?, I'm using companion - I don't own a radio

3djc, 2020-04-05:
Sorry mistyped : You can calibrate withIN companion simu.. I'm having a look at a possible callibration issue in the simu

myozone, 2020-04-05:
Ah ok, I'll have a play - still learning

myozone, 2020-04-05:
I get it to work once but it stopped again, also the '1' is greyed out a bit. It was lighter when it worked.
'Calibrated' a few times and re-selected  Mutlipos Switch for 6P within Hardware.

kilrah, 2020-04-05:
The number staying at 1 is a known issue but the rest should work, even without calibration

myozone, 2020-04-05:
Good to know, basically then the 6P mode switch in the simulator doesn't work. Everything else works apart from that.

3djc, 2020-04-05:
It mostly work, but as on the radio, it requires calibration. The calibration on the simu as sommes issues, but if getting à profile from a real radio, it works fine

myozone, 2020-04-05:
Ok, I'll wait till I get a real radio - unless you have a real radio profile ?