opentx/opentx #7496: x9d+2019 open tx 2.3.7 doesnt work

bandit1928, 2020-04-05:
i flashed it now my radio is dead

3djc, 2020-04-05:
You cannot brick a taranis. You probably flashed the wrong firmware, but this site is not an end user support site, but a bug tracker. Please go to community places like rcgroups or openrcforum for end user support

closed 2020-04-05 by 3djc

susomartinezromero, 2020-05-05:
Hi! i have the same problems, flashing with 2.3.7 doesnt boot, 

I have tested sometimes with the same result,  i can not boot it. Nonething on the screen.

I have rolled back to 2.3.5 and work perfectly.  My radio is taranis x9d+ Especial Edition 2019.

I thinks that its not a end user problem. Thanks for all!

3djc, 2020-05-05:
what is the exact name of the firmware you are trying to flash ?

susomartinezromero, 2020-05-05:
this one (not boot):

opentx-x9d+2019-flexr9m-lua-es-2.3.7.bin

The rollback (working perfectly) is to:
tbs_firm_x9d+2019_en_20200106.bin
To use Crossfire shot.

3djc, 2020-05-05:
Well then try one of our own first

kilrah, 2020-05-05:
> this one (not boot):
opentx-x9d+2019-flexr9m-lua-es-2.3.7.bin

Just tried that on my X9D+2019 and it works fine, so not an issue here.
Try flashing it differently, e.g. from SD card or with power off.