opentx/opentx #7497: Remember scroll position in all model menus

kniuk, 2020-04-05:
Regarding Horus / T16 color LCDs:

IMHO it would be great if when in Model Menus switching to different Page (model setup, flight modes,  inputs, etc..) vertical scrolling position would be remembered. That would greatly facilitate setup. Switching between pages now always resets the scrolling position to the top.

kilrah, 2020-04-05:
There is no relation whatsoever between the vertical position between screens so it doesn't make much sense, would just be confusing.

kniuk, 2020-04-05:
example:
1. editing logial switches L34
2. page up to mixes on CH12
3. back to logic switches tweaking L34 behaviour
4. back to mixes on CH12 adding final tweaks

Now all those operations require scrolling from CH1 on mixes and from L01 on logical switches.
My wish would be that every page has its scroll posiiton remembered so somebody doesn't have to scroll down to find last edited values. Exiting to main screen can reset those scrolling positions, so as you said no to be confusing after entering menus for the second time.

kilrah, 2020-04-06:
Ah, for each page. Yes that makes a bit more sense

3djc, 2020-04-06:
the little difficulty here is that menus are dynanic, so by changing a value in a screen, you might affect the number of lines in another, so there might be some issues with that. In 2.4, colorlcd UI is quite similar, but quite different too in its mechanics, will have to see the impact of that

kniuk, 2020-04-06:
Even if you a user would need to scroll a little bit, it would be better than scrolling from the top, however I understand that you guys seek for perfect solutions, otherwise openTX wouldn't be where it is now :)

kniuk, 2020-10-16:
3djc have you managed to find some time to look into that ?

d-hartmann, 2020-12-26:
I also don't see a case here. Yes, it would be nice in some cases, but mostly, I'm very sure this will only confuse. There should also be a timer (restart or seconds/minutes) reset the sroll position.
Maybe the thing is not to remember the scroll position, but scroll for a more than just one element.
Maybe the solution would be to scroll be 1 without an additional button and scrolling by 10 with another. So you can easily jump over many items at once, as you can jump back with long pressing.

closed 2021-10-02 by stale[bot]