opentx/opentx #7499: board_common.h:144: multiple definition of `ticksNow'

gretel, 2020-04-05:
trying to compile branch `2.4` from today:

```shell
docker run --rm -it -v "$PWD:/opentx" -e "CMAKE_FLAGS=HELI=NO FLIGHT_MODES=NO GVARS=NO SPLASH=NO JITTER_MEASURE=YES HARDWARE_TRAINER_MULTI=YES" -e "BOARD_NAME=x9lite" vitass/opentx-fw-build
```

```
[ 32%] Linking CXX executable bootloader.elf
CMakeFiles/bootloader.dir/__/usbd_desc.c.o: In function `ticksNow':
/tmp/opentx/radio/src/targets/common/arm/stm32/board_common.h:144: multiple definition of `ticksNow'
CMakeFiles/bootloader.dir/__/__/__/__/__/thirdparty/STM32_USB-Host-Device_Lib_V2.2.0/Libraries/STM32_USB_OTG_Driver/src/usb_dcd_int.c.o:/tmp/opentx/radio/src/targets/common/arm/stm32/board_common.h:144: first defined here
CMakeFiles/bootloader.dir/__/usb_bsp.c.o: In function `ticksNow':
/tmp/opentx/radio/src/targets/common/arm/stm32/board_common.h:144: multiple definition of `ticksNow'
CMakeFiles/bootloader.dir/__/__/__/__/__/thirdparty/STM32_USB-Host-Device_Lib_V2.2.0/Libraries/STM32_USB_OTG_Driver/src/usb_dcd_int.c.o:/tmp/opentx/radio/src/targets/common/arm/stm32/board_common.h:144: first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [radio/src/targets/common/arm/stm32/bootloader/CMakeFiles/bootloader.dir/build.make:736: radio/src/targets/common/arm/stm32/bootloader/bootloader.elf] Error 1
make[2]: *** [CMakeFiles/Makefile2:1317: radio/src/targets/common/arm/stm32/bootloader/CMakeFiles/bootloader.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:286: radio/src/CMakeFiles/firmware.dir/rule] Error 2
make: *** [Makefile:251: firmware] Error 2

ERROR: make compilation failed.
```

tried to figure but did not :(

3djc, 2020-04-05:
2.4 is not ready at all

closed 2020-04-05 by 3djc

gretel, 2020-04-05:
@3djc this is the spirit. thank you