opentx/opentx #7501: Custom sensor instance changed by upgrade from 2.2 to 2.3

jfrickmann, 2020-04-07:
Most models in my SoarOTX package have a custom sensor with ID 5050 and instance 0. It is written to by a Lua function script, and I use it for logging the current flight mode, to assist the log graphing script separating flights from one another.
When I upgraded to 2.3.7, the instance was changed to 32. I can set the instance back to 0 in Companion (and tell all of my users to do the same) but not on the radio. On the X9D+ radio, the "ID" on the TELEMETRY screen shows 255. When I edit the sensor on the radio, it shows 5050 and 32, but I cannot change 32 to 0; it stops at 1.

jfrickmann, 2020-04-07:
UPDATE: the above fix does not work on the radio, only in Companion. I will now try to change the instance number used in the script and see if I can get it working again :(

3djc, 2020-04-07:
Because of access, there is a new way to handle instance on the radio that includes the source module/receiver

jfrickmann, 2020-04-07:
How do I get the lua sensor working again?

jfrickmann, 2020-04-07:
UPDATE: When I set the instance to 32 in Companion, it shows ID 255 on the TELEMETRY screen. When I use instance=255 in setTelemetryValue in Lua, it works in Companion.
I feel that you need to clarify the differences between ID, sub ID, and instance, and use the terminology consistently in Companion, on the radio screen, and in the Lua Reference guide.

jfrickmann, 2020-04-07:
Since it works on my transmitter with the new numbers, I will close this issue. Please update the documentation to explain how instance #32 on Companion becomes ID #255 on the radio.
And thanks again for all of your hard work!!

closed 2020-04-07 by jfrickmann

jfrickmann, 2020-04-07:
When I wanted to update my templates for GitHub with the fix, I discovered that they were unaffected by the issue. Apparently, it was only models that had been loaded on my radio. I am guessing that it somehow happened as I was upgrading from 2.2.4 to 2.3.5, back to 2.2.4, and then to 2.3.7. So probably a very fringe case!