opentx/opentx #7506: Jumper T16 backup model to SD Card via radio context menu as per QX7

davidabevan, 2020-04-09:
Cant see any option  to enable a quick backup of model to sd card yet, nor any bug reports.
As per QX7 hold  select button for context menu then selecting backup model to sd card
I find this option really useful when adapting models being able to backup on the fly.
Would be grateful if could be added to enhancement for T16 pro v²
Thanks..

3djc, 2020-04-09:
You can just copy a model from the model selector

davidabevan, 2020-04-09:
Thanks yes agreed you can, but it's not the same Functionality. All the
other open tx radios to my knowledge have this function that is very
useful. Eg if your model memory overflows you can safely delete them
knowing that on the field you can restore a particular model again if
needed. There are a number of ways I use it. Also if I do duplicate models
on the fly then the memory shows too many models leading to error possibly.
Much easier to have that functionality as per request.
Many thanks for the suggestion.
Thanks..

3djc, 2020-04-09:
No such limit on Horus. You can also copy paste a model from /MODELS to where you want using the SD manager, plenty of ways

davidabevan, 2020-04-09:
Yes I can look into that. The T16 is somewhat different to the qx7 nirvana
and xlite. These I own and all have the sd card backup functionality. I do
hope that the devs will consider the request. But yes I am working around
it. I just find it very useful to hold that button and say backup to sd
card, before I start changing usually my logical scripts. I will however
look into your suggestion.
Thanks.

3djc, 2020-04-09:
That fonctionnaly exist on Taranis because the mdoel are stored in Eeprom, and are therefore not accessible. On Horus/T16/TX16s, all the model are already files on sd card

davidabevan, 2020-04-09:
Okay.. Thanks Guess I don't understand my new radio infrastructure so to
speak properly yet. I will experiment to learn. I believe you may be a
dev.. If so thanks for great work.

davidabevan, 2020-04-09:
Function already exists just implementation is different to what i am used to due to different radio memory structure

closed 2020-04-09 by davidabevan

3djc, 2020-04-09:
Just to help you understand how it is on Hoprus/T16/TX16s

/MODELS/modelX.bin : indiviudal model files. You can have more files here than appear in your model selector (see bellow models.txt)
/RADIO/radio.bin : contains your radio settings (stick mode for exemple)
/RADIO/models.txt : maps individual model files within categories. Only 'active' models are registered here. When you delete a plane from model selector for exemple, only it's index in models.txt is deleted

davidabevan, 2020-04-09:
> Just to help you understand how it is on Horus/T16/TX16s

Really useful tips there. Thanks very much. I have just started
experimenting and am beginning to see what you mean. Have created a
category called backup and moved my newly converted working models to it.
Will investigate as you suggested.
Many thanks for your help.
Dave..👍😁