opentx/opentx #7509: Windows size companion 2.2.x and 2.3.x under Linux

nostromo123, 2020-04-09:
I think a see an issue here with oTx 2.2,.4 and 2.3.7 on my Thinkpad T450s running DEBIAN Testing.
Max. screen resolution is set to 1366x768

Full screen starting 2.3.7
[image]

and this is a 'full' screen with model setting opened
[image]

The screen size of that window can not be reduced. Reduce window size is grayed out.

Not usable ...

br KH

jtaylor2, 2020-04-09:
Clicking on a window edge or corner and dragging the window smaller doesn't work?

nostromo123, 2020-04-09:
No.
I am running a VM on that T450s with WIN 10. (1366x768)
Here the fullscreen, which is ok.
[image]
 And here my desktop PC (DEBIAN Buster 1680x1050) with min size for that window.
Can reduce height to that min, but you are seeing min width.
[image]

br KH

nostromo123, 2020-04-10:
Seems to be a DEBIAN issue.
Installed Linux Mint 19.3 Xfce on the T450s.

There it looks ok.

[image]

I am going to discuss that in DEBIAN forum.

Sorry and br - KH

closed 2020-04-10 by nostromo123

jtaylor2, 2020-04-10:
I tried yesterday to duplicate your problem in my self compiled openSUSE Leap 15.1 and Windows10 systems and could not duplicate it at any of my available screen resolutions.

For what it's worth I loaded Debian 10 in a virtual machine and tested the .deb at a screen size of 1360x768 and saw no problem.
[image]

nostromo123, 2020-04-10:
Thank you for trying, Jim. The T450s is on Testing, I guess, with Buster (DEBIAN 10) it could be ok.

I also asked in the Xfce forum and got this answer. [https://forum.xfce.org/viewtopic.php?id=13894](url)

So seems to be NOT an oTx and NOT a Xfce issue, it is pointing in the DEBIAN Testing direction.

Best regards and have a nice day - KH