opentx/opentx #7510: trainer signal lost on T16 - opentx 2.3.7

JCPDE74, 2020-04-09:
"Trainer signal lost" being triggered multiple times in a few minutes. Cannot pinpoint what is triggering it. Release 2.3.7 Thanks for investigating it.

kilrah, 2020-04-09:
Can't investigate much without knowing what you said the trainer settings to.

JCPDE74, 2020-04-10:
> Can't investigate much without knowing what you set the trainer settings

Hi Andre. I did not played with anything. Sd Card zeroed out, firmware
flashed and voilá... "trainer signal lost" started to be triggered.

JCPDE74, 2020-04-10:
I'm using the radio on a simulator. Internal / external modules off (stock configuration)

3djc, 2020-04-10:
Then why did you setup trainer on this model ? Please join otx and specify which model you are using

JCPDE74, 2020-04-11:
That is the thing... I did not set up trainer on this model. 
I was looking how to disable it and found that on the Jumper T16 (I guess in other radios too), there are 3 modes for trainer - master/jack, slave/jack,master/multi, but there is not a "off" selection available.

JCPDE74, 2020-04-11:
Thank you for helping me. I appreciate your efforts

kilrah, 2020-04-11:
If you set to slave it will output a signal and thus not check for an incoming one...

JCPDE74, 2020-04-11:
Thank you Andre. I followed you recommendation, made a couple of flight and
it is working without warnings to this moment.

Maybe this should be the default setting when loading the software on the
radio.

closed 2021-01-09 by stale[bot]

Nihonski, 2021-05-19:
check this: https://github.com/opentx/opentx/issues/8513
There is a signal conflict/crash on the heartbeat signal

Cpt-Huck, 2022-06-18:
FlySky Nirvana 14 (NV14) - EdgeTX 2.7.1
just had this when writing an Backup via EdgeTX Configurator to my Radio.
If I go to "Model Setup" - scroll all the way down - and on "Trainer" - "Mode" - "Master Jack" is the default I guess ?!
Gives me the same problem you have "Trainer Signal Lost / Recovered". (Random ?! and when navigating through the menu ?!)
"Master/Multi" = no problems. But I just turned it "OFF" since im not using it. = Problem solved / fixed with this Setting.

Nihonski, 2022-06-18:
If this issue is not solved yet, the HW might get damaged, as the Heartbeat signal driver shorts the SBUS signal, when the timing matches.
[image]
.