opentx/opentx #7515: Q X7 OTX >= 2.3.5 Double press powerbutton sometimes render device dead

Fumes74, 2020-04-12:
If I double tap the powerbutton during startup it sometimes render radio "bricked".
Meaning I can't turn it on anymore. A clicking sound from speaker and a very short blink of the backlight is all I get. Leaving radio with battery disconnected doesn't help. Even removed internal storage battery overnight without results. Only remedy is to flash firmware via DFU. All models and settings are left untouched as I can see it and radio works as intended.

Frsky Q X7.
OpenTX 2.3.5 and above.
OpenTX Companion 2.3.5 and above on Windows 10.

Fumes74, 2020-04-12:
I'm not so sure it's a bug anymore. Could be just my radio being flaky, maybe a bad eprom or other memory fault. Now it happens sporadicaly when i turn it on "normaly" without double press.

d2f2, 2020-06-02:
Check rtc battery and mount new SD card for test.

closed 2020-11-29 by kilrah