opentx/opentx #7521: Random change of GV9 value

afdofdez, 2020-04-14:
Horus X10S Expres -- Opentx 2.3.7 -- EU/LBT.

Since I bought my Horus X10S radio, and installed OpenTx in the first version that supported it, until today I have version 2.3.7 in the models that use all the local variables GV1-GV9, sporadically, today He has been configuring the telemetry screens but other times doing other things, he only changes the value of GV9 to 1%, I usually have 80% and it is the dual rate of the ailerons, I hope he never does it to me in flight.

In one of the models that used GV1-GV9 I stopped using GV9 and it never gave any more problems, in another I still use GV1-GV9 and the problem persists, I see it every 10-15 times I use the model.

It gives the impression as if the memory size of the GV`s was not well dimensioned or not.

kilrah, 2020-04-14:
Can you supply a model where the issue is reproducible?

afdofdez, 2020-04-14:
Cheers

Enviado el: martes, 14 de abril de 2020 13:33
Para: opentx/opentx <opentx@noreply.github.com>
CC: Fernando Fernández <a.fernando.fernandez@gmail.com>; Author <author@noreply.github.com>

kilrah, 2020-04-14:
If there was supposed to be an attachment, github doesn't take email attachments.

afdofdez, 2020-04-14:
Modelo Heron.otx     https://drive.google.com/drive/u/0/my-drive

Enviado el: martes, 14 de abril de 2020 14:11

kilrah, 2020-04-14:
Wrong link.

afdofdez, 2020-04-14:
https://drive.google.com/open?id=1KajJosMmUXWc5g1HYGe1vO2_vWKAQZ7t

 

Sorry

Enviado el: martes, 14 de abril de 2020 18:40

robustini, 2020-05-24:
I can confimr the bug.
I had models that used GV9 and the first value I had set it to 88, now it is 1.
Reproducing the bug is simple, just change that first value from tx, exit and re-enter the page, go back to 1.

robustini, 2020-05-24:
I can confimr the bug.
I had models that used GV9 and the first value I had set it to 88, now it is 1.
Reproducing the bug is simple, just change that first value from tx, exit and re-enter the page, go back to 1.
Here only change the first value of GV9, not the others.

robustini, 2020-05-24:
A video showing the bug:

https://youtu.be/-0c10dJ24lw

3djc, 2020-05-24:
Can’t reproduce here with tx16s, does the same happen on a empty model without yapuu and all ?

afdofdez, 2020-05-24:
I found the video of the GV9 failure very interesting where it is observed that the model has Yapu telemetry.
I also have the Yapu telemetry installed on another model, not the one that uses the variables GV1 to GV9.
I wonder, is it not an error of the yapu telemetry, which overwrites the GV9 value of some model to 1?

kilrah, 2020-05-24:
A script cannot touch anything but the current model.

robustini, 2020-05-24:
Don't ask me why but I just created a copy of that model by removing the Yaapu and that variable now doesn't change it.
I try to investigate with Alessandro Santilli, the author of Yaapu if there can be a relationship.

StefanKellerAC, 2020-05-25:
I can confirm it: 
The yaapu mapping widget uses GV9 FM0 and FM7

3djc, 2020-05-25:
So nothing wrong with OpenTX then, right ?

robustini, 2020-05-25:
Yes, nothing wrong with OpenTX.

3djc, 2020-05-25:
Ok, closing this then.

closed 2020-05-25 by 3djc