opentx/opentx #7522: Owner Registration ID in companion does not allow numerical characters

plogibear, 2020-04-15:
Hi,

I found a bug in OpenTX companion V2.3.7 its connected to an FrSKY Xlite Pro.
Under Radio Settings >> Owner Registration ID (Owner Registration ID in companion does not allow numerical characters).

However on the FrSKY Xlite Pro you are able to change your Owner Registration ID it accepts both alphanumerical characters.

[image]

closed 2020-04-21 by 3djc

pr-newman, 2020-04-25:
Issue also applies to FrSky Taranis X9D+ 2019 with v2.3.7.  Same details as for FrSKY Xlite Pro.

3djc, 2020-04-25:
It has been fixed already, please get a nightly companion

pr-newman, 2020-04-25:
Thanks - I twigged it was Companion not Firmware after I commented! 8-)  Cheers --Pete

 

From: 3djc [mailto:notifications@github.com] 
Sent: Saturday, 25 April 2020 15:32
To: opentx/opentx
Cc: pr-newman; Comment
Subject: Re: [opentx/opentx] Owner Registration ID in companion does not allow numerical characters (#7522)

 

It has been fixed already, please get a nightly companion

PR #7523 (2020-04-21): Allow alphanumerical in ownerID
This fixes #7522