opentx/opentx #7525: Linux version of companion app as appimage

fajabird, 2020-04-16:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
not really - only that today there are more effective ways to offer applications for linux users (see JITSI project here https://github.com/jitsi/jitsi-meet-electron/releases that offers appimage releases for download)

**Describe the solution you'd like**
Instead of building debian/ubuntu specifig DEB-packages just switch to appimage. This files can just be downloaded and started on most linux distrubutions without any installation. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

padcom, 2020-12-25:
That'd be awesome! +1

olexam, 2021-01-20:
Hi, using Open SuSE, so unable to use Companion app directly, really need cross platform binaries

closed 2022-04-17 by stale[bot]