opentx/opentx #7526: OPCIÓN MULTI MÓDULO EN VERSIONES POSTERIORES A 2.2.4

lauchablanca, 2020-04-17:
Hola gente Mi QX7 tenía la versión 2.3.0, pero no tenía la opción MULTI Módulo en esa versión, tampoco en la descarga de COMPANION más actual.
Tuve que hacer un DOWNGRADE, a la versión 2.2.4, para poder elegir la opción MULTIMÓDULO en el Companion.
¿Por qué desapareció esa opción desde la versión 2.2.4 en adelante?
No se puede agregar a versiones posteriores a 2.3.0?
Para poder usar mi Módulo externo IRX4 tengo que utilizar una versión vieja de OPENTX.

kilrah, 2020-04-17:
Multimodule is always included since 2.3.9.

closed 2020-04-17 by kilrah

lauchablanca, 2020-04-17:
[image]

lauchablanca, 2020-04-17:
Where is the multimodule option?

3djc, 2020-04-17:
Please read above ....

lauchablanca, 2020-04-17:
> Por favor lea arriba ...

Friend, I read, but  I download the OpenTX Companion more recent that 2.2.4, and not have the multimodule option. Do you see my shared picture?

3djc, 2020-04-17:
READ THE ABOVE POST BY KILRAH !

lauchablanca, 2020-04-17:
> ¡LEA EL ANTES POSTAL DE KILRAH!

Ahh, ok ok. I will read now. Thanks

lauchablanca, 2020-04-17:
Friend, I don't want to bother or be pushy. Does the multi-module option appear with another name? They just tell me "it's always included since 2.3.0", but I installed at least 5 or 6 Comanion yesterday and none had it.

3djc, 2020-04-17:
....

ANY 2.3.x firmware DOES include it, whithout the need to tick any option

lauchablanca, 2020-04-17:
Ahh, oK. Thanks. I will try now. Thanks again