opentx/opentx #7527: Add option for calculated sensor

franz4949, 2020-04-17:
With ACCESS we now have the additional FLR sensor with unit %. A value of 0% means no frames lost, means excellent link quality, a value of 100% means all frames lost ansd probably causes FS. A new value (let me call it LQ) could be calculated from the FLR value by LQ = 100%-FLR. That would allow to correlate good link quality with a high percentage value, which we users are accustomed to see e.g. for RSSI and also for the many times shown and discussed Tadango LF sensor values on RCG. The actual sensor calculation allows adding several sensor values including its negative values but it does not offer an option to add a constant value (here e.g. needed 100%). So my proposal is to extent the adding list for a selectable constant.

3djc, 2020-04-17:
can already be done through LUA

franz4949, 2020-04-17:
LUA unfortunately is not for all ... could be so easy with my proposal, maybe you reconsider.

franz4949, 2020-04-17:
Will leave it up to you, so you may close it if it doesn't fit into your planning.

franz4949, 2020-04-24:
FrSky will make the conversion in its new ACCESS Rx firmware, so I close this request.

closed 2020-04-24 by franz4949