opentx/opentx #7534: Jumper T12 Pro losing signal when doing a voice countdown

CurryKitten, 2020-04-19:
A user commented on a YouTube video I made about the T12 Pro that he was getting signal loss after he'd implemented the voice countdown.  It was at a set point in the countdown, he'd got it at the 30 seconds warning, the 10 seconds and part way through the countdown.  Even after setting the countdown to beep, he'd lost signal on the "Timer 1 elapsed" sample.

He was running 2.3.7 of OpenTX with the SD card image from the OpenTX site and was using an XM+ in his quad.  Thinking it might have been an SD card issue, he tried both reformatted and using a different card but had the same issue.

I attempted to recreate the problem (albeit on OpenTX 2.3.4).  I used the regular SD card image from the OpenTX site firstly on a 2GB unbranded SD card - this had previously been working fine in a FrSky QX7.  I was able to replicate the problem.  I added a timer for 1 minute based on THs and changed the countdown to voice.

[image]

I hovered an SPI based whoop quad and on the first occasion it fell out of the sky at the 10 second warning, and then on the next test, the countdown got to 5 before it lost signal.  During this time I also noticed that the voice samples could be a bit scratchy/have break up, and the countdown would occasionally miss out a number i.e 9,8, (pause) 6, 5 etc

I retested using a known good 32GB class 10 SD card, but had the same issue - again losing signal on the 10 second countdown - although one time I got the voice message "receiver still connected" which I'd normally only expect to hear if I tried to turn the radio off when still connected.

As far as I can tell, this appears to be unique to the T12 Pro.  I'm using FrSky X9D, QX7, X-Lite & X9 Lite Pro with similar levels of OpenTX and have never had any sort of issues like this.

3djc, 2020-04-20:
Please try nightlies, we fixed an issue that might be related

CurryKitten, 2020-04-20:
I installed 2.3.7 just to confirm the problem was there for me as well (it was) so I went on and installed 2.3.8N189.  I did several static and more hover tests and saw that the issue went away.  The voice samples play fine and there's no interruption in the signal.  I did have a look for any issues that might have looked like this - but I didn't find any.  Which issue was it... and I presume it'll go into the eventual release of 2.3.8?

kilrah, 2020-04-20:
It was a DMA conflict between module and SD card. https://github.com/opentx/opentx/pull/7505

closed 2020-05-08 by 3djc