opentx/opentx #7536: Make global variables selectable as curve values

RealTadango, 2020-04-20:
For my heli i use a flat curve but the min and max pitch depend on the flight mode. I have used the throttle value (for govenor) as a GV which depends on the flight mode so i only need 1 mixer line there. It would be great if i can use 1 curve and set the curve values (bottom line) with GV's. Then the curve can be adjusted with GV's by setting them per FM. There are other cases for using GV as CV inputs also. It would be an awesome feature :)

RealTadango, 2020-10-20:
Still wanted

padcom, 2020-12-25:
Actually I'd like to see that feature too. It'd be very helpful in setting up slider-driven overrides for channels, for example for the elevator on long climbs/descents when doing long-range flights. I have for now solved it with a bunch of things:

- creating an input that takes a pot and scales it to match the max deflection
- storing that scaled value in a GV (needed because the weight cannot be taken from an input)
- doing a set of overrides based on the value of a switch and elevator side (below 0 and above 0)

The process is extremely cumbersome but could be made very easy just by editing the center point of a 3-point curve and assigning that to the channel...

padcom, 2021-01-24:
I just found another way to do the same. Essentially I want to have the option to not have to keep on pulling or pushing the elevator to climb because that is not accurate and the thumb just hurts after a few minutes.

1. create two inputs (one for climbing, one for descending) that output values from 0..whatever (I set mine to weight 40 offset 40 for descent and weight -40 and offset -40); I used a slider for it, but you can use a pot too.
2. create three special functions:
   - one for setting a global variable from first input created above
   - one for setting a global variable from the second input above
   - one for setting the global variable to zero (level flight)
3. Now here is where the magic goes: I have set the trim on Output tabs to that global variable and disabled "linear" scale (last checkbox in the output line; most probably not enabled by default)

For it to work as intended I needed to properly recalibrate the sticks, pots and sliders so that the output in the mid position of the sticks is exactly 1500 or else the flight controller would go nuts.

Even though it is a much simpler method it is way less intuitive, and I found it by accident. Didn't know the trim could do that. That being said I still wish the curves would be parameterized - at least the middle point of the 3 point curve, best with an input, but a GV would be enough.

closed 2022-04-17 by stale[bot]