opentx/opentx #7539: Possible error in calculating the% in the battery script

afdofdez, 2020-04-21:
**Horus X10S Express V2.3.7 EU** 

Hi, I'm not sure it's a problem, but it seems to me.
In the last two or three versions I have seen that improvements have been made in the battery script, however the value of the battery in % that is presented every time is smaller and moving away from the battery value that my CallMeter 8 presents. , which I use as a pattern.
I have verified that the voltage values of the cells and the battery are the same that the FLVSS presents, however the calculation in % of battery capacity has been moving away (decreasing) version to version of the one indicated by my meter. I am aware that my meter can do it wrong, but it certainly does as well or as bad today as it does three versions.

Could you check the calculation of the % of the battery?

3djc, 2020-04-21:
The same formula seems used when compared to 2.2 scripts 3 years ago. Could you provide an example with all cell values ?

afdofdez, 2020-04-21:
BATTERY "A" 3S 2200 mAh

1S

2S

3S

TOTAL

%

cell meter 8 hw

3,83

3,84

3,87

11,54

48

FVLSS

3,82

3,85

3,86

11,53

 

SCRIPT BATT

3,82

3,85

3,86

11,53

41

						

BATTERY "B" 3S 2200 mAh

1S

2S

3S

TOTAL

%

cell meter 8 hw

3,9

3,87

3,91

11,69

59

FVLSS

3,9

3,88

3,9

11,68

 

SCRIPT BATT

3,9

3,88

3,9

11,68

58

						

BATTERY "C" 3S 2200 mAh

1S

2S

3S

TOTAL

%

cell meter 8 hw

3,86

3,82

3,86

11,56

49

FVLSS

3,86

3,83

3,5

11,54

 

SCRIPT BATT

3,86

3,83

3,5

11,54

44

						

BATTERY "D" 3S 2200 mAh

1S

2S

3S

TOTAL

%

cell meter 8 hw

3,85

3,83

3,86

11,55

48

FVLSS

3,85

3,83

3,86

11,54

 

SCRIPT BATT

3,85

3,83

3,86

11,54

44

 

THE DIFFERENCE I HAVE OBSERVED BETWEEN VERSIONS 2.3.X

Enviado el: martes, 21 de abril de 2020 15:52
Para: opentx/opentx <opentx@noreply.github.com>
CC: Fernando Fernández <a.fernando.fernandez@gmail.com>; Author <author@noreply.github.com>

3djc, 2020-04-21:
@offer-shmuely could you have a look at the lastest versions of the script
Today :
[image]

May 2019 version
[image]

maru-sama, 2020-04-21:
Hi,

I had a quick look at the code. Apparently in the newer version the percentage is calculated based on the minimum Cell value while in the old version it is calculated based on the averageValue of all cells.

afdofdez, 2020-04-21:
My opinion is only based on practice and day to day, it seems important to me to obtain similar values in the software (script battery) and in the hardware (CellMeter), otherwise confusion and lack of confidence in the user may be created, which you will always end up relying on the hardware meter.

 

It's just an opinion.

Enviado el: martes, 21 de abril de 2020 19:19

kilrah, 2020-04-21:
> My opinion is only based on practice and day to day, it seems important to me to obtain similar values in the software (script battery) and in the hardware (CellMeter), otherwise confusion and lack of confidence in the user may be created, which you will always end up relying on the hardware meter.

There are at least 4 different voltage to % curves around, so if you take 2 different hardware meters you're likely to get different results as well.

afdofdez, 2020-04-21:
Wouldn't an option in the battery script be possible to configure the curve?

Enviado el: martes, 21 de abril de 2020 21:25

kilrah, 2020-04-21:
Don't think it makes sense over editing it.

offer-shmuely, 2020-04-22:
I think that it depends on the perspective you are looking on the battery.

lets take this case to extreme (just for the test)
if you have 2 cell battery, with 6.7v, one cell is 3.0v (0%) and the other is 3.7v (14%)
The question here, **_can you use the battery for additional time?_**
I think the answer is "no", 
so the battery can be consider as **0%** even that one of the cell have more juice (no, I don't run the battery so low)

however, if are looking at the **total** voltage of the battery, then you have 6.7v, and that can be consider 6%.

Maybe I can add additional display field (in small fonts) that show the average percent as well, so there will be a reference to field-meters (assuming that's how they calculate it).
But that is possible mainly on the big widget (the full screen one), and not on the smaller ones.

what do you think?

[image]

afdofdez, 2020-04-22:
I think that filling the screen with small characters that are not appreciated during the flight does not make sense.

It seems better to me to add a second parameter to the widget (cells, range) where the value of range "medium" or "minimum" determines the formula to use in calculating the%.

The presentation would only be modified with a symbol after the% that indicates whether it is presenting the "average" or "minimum" value.

 

Thank you

Enviado el: miércoles, 22 de abril de 2020 22:06

 

 <https://user-images.githubusercontent.com/7026911/80028383-b959b100-84ed-11ea-9724-7fc6957493a8.png>

kilrah, 2020-04-22:
IMO percentage should be calculated with total voltage, remember that it's only good at rest in the first place, i.e. no load on the battery.
Lowest cell too low warning should be done with a logical witch based alarm for example.

3djc, 2020-04-23:
It was designed from scratch to match the lipo tester that all of us probably have in our toolbox. I'm sorry I did not spot the change in calculation rule, because I think it is wrong, it should show what your tester show, and testers use average value, and as Kilrah said, alarms should be done with lowest cell calculated sensor (and not use LUA that is not available in emergency mode)

offer-shmuely, 2020-04-24:
> remember that it's only good at rest in the first place, i.e. no load on the battery

That gave me an idea, since some of us (me) use it for inflight monitoring (i.e. under load), 
and for that, there should be a different mapping of the V-->% mapping anyway,
I can add "in-flight / rest" parameter, so:

- rest mode (or no load): 
  -  calculating by average voltage
  - as similar as can be to a field tester
  - with the current mapping of V--> %

- flight-mode (in load): 
  - calculating by min-cell voltage, 
  - different mapping of V-->% that represent better the % on load

what do you think?
do you have any idea for name to such a parameter?

3djc, 2020-04-24:
LUA should NOT be used for mission critical stuff, which your lipo alarm is

offer-shmuely, 2020-04-24:
it's not.
I have 2 alarms (low and critical) for each of the two battery!
This is just for a reference, how good is the condition of the battery during vertical climb, and how match more flight time I can expect.

kilrah, 2020-04-24:
Calculating % by min cell voltage is never good IMO.
A compensation for load can work but is complex to put in place, every single battery will need some calibration etc so I'd say it has no place in the default script we ship with OpenTX. Should be a 3rd party add on that people can download and go through implementing if they want.

offer-shmuely, 2020-04-24:
Will be fixed soon.
(Together with reduce usage from 70-90% to 25%)

offer-shmuely, 2020-04-28:
I like to open the discussion again.
I check some batteries with my field tester (I consider the ISDT as the best one)
and this is my finding:

A battery that is not so balance on ISDT-BattGo :
[image]

BattCheck widget coded with the cell average (cell-avg = 3.78v) 
opentx: 30% (field-tester: 16%)
[image]

BattCheck widget coded with the cell minimum (cell-min = 3.71v) 
opentx: 15% (field-tester: 16%)
[image]

So, for conclusion, I think the calculation should be done by cell-min
what do you say?

kilrah, 2020-04-28:
That calculating via cell min makes no sense. It's not because one cell is lower that the pack is that much lower overall.
The fact that a cell is much lower is bad, but reporting it as a reduction in percentage means nothing. The dV value is there for that and you'd normally set up an alarm on that. If the alarm sounds you come land immediately, the percentage is irrelelvant.

offer-shmuely, 2020-05-11:
any comment?
do we like the widget to be as close as possible to a field meter?
what type of field-meter? (ISDN / Futaba / Other)

3djc, 2020-05-11:
The reference value comes from my Robbe one (unless you changed them at one point in time), so I guess this is the reference one

3djc, 2020-08-12:
Fixed in last release

closed 2020-08-12 by 3djc