opentx/opentx #7545: Companion 2.3 Hangs after closing simulator (Jumper T16 Ubuntu 18.04)

JMare, 2020-04-22:
**Describe the bug**
After closing the simulator, companion becomes unresponsive and has to be killed.
Tested with 2.3.7 stable and 2.3.8N191 on Ubuntu 18.04. SD contents are unchanged.
Problem does not occur on windows. The simulator is functional before you close it.

**To Reproduce**
Steps to reproduce the behavior:
- Open companion23 on ubuntu 18.04, configure for jumper t16
- Create blank model using wizard (.otx file attached)
- Open simulator
- close simulator
- Window will be unresponsive

**Expected behavior**

Companion remains usable after the simulator is closed

**General information**
 - OpenTX version 2.3.7 and 2.3.8 N191
 - Hardware Jumper T16
 - OpenTX Companion 2.3.7 and 2.3.8 N191

See opentx file:
[opentxtestmodel.zip](https://github.com/opentx/opentx/files/4519309/opentxtestmodel.zip)

Companion23 output:
[companion238output.txt](https://github.com/opentx/opentx/files/4519312/companion238output.txt)

3djc, 2020-04-23:
Working fine on debian strech here

JMare, 2020-08-11:
Well I went back and tried this again with 2.3.9, couldn't reproduce the issue with everything else the same, so closing this.

closed 2020-08-11 by JMare