opentx/opentx #7546: Jumper T16 Pro Hall

Joshenriquei, 2020-04-23:
I bought a new Jumper T16 Pro Hall, direct from China, in Ali express site. This was in December 2019. I received the radio on February 2020, about first week. After using the transmitter for two weeks without problem, I get an issue, the button page stops to scroll over the top menu. I get the beep, but the cursor do not move on menu. Then, was impossible to program the model.
Then I change to OpenTX. Just now I am using OpenTX 2,3.7 and sdcard-480x272-2.3V0026.zip. But, without any menu on screen.
Is there some procedure to solve this problem?
The transmitter continue to work, the SD Card is OK, I can create a new model, bind the receiver, but nothing about programming.
I know about the fight between OpenTX and JumperTX , and I am with OpenTX.
 Some help?

kilrah, 2020-04-23:
Problems with the controls are due to the ribbon cables inside the radio, remove/clean/reseat them

closed 2020-04-23 by kilrah

Joshenriquei, 2020-04-23:
Sorry Mr, but the problem is not control. The issue is in program. I already disconnected the flat cable between SDCard support and main board, clean it and connect again.  The issue continue. Someone have a new suggestion?

3djc, 2020-04-23:
Maybe do a video ? Really hard to understand why you are talking about

kilrah, 2020-04-23:
The ones you need to reseat are those that go to the button and the wheel.

Joshenriquei, 2020-04-23:
[image]
This is the problem.

3djc, 2020-04-23:
Your sdcard content is wrong

Joshenriquei, 2020-04-23:
[image]
This is my SDCard contend.

3djc, 2020-04-23:
Very wrong indeed. You need to bring up he content of the sdcard folder to sd card root

kilrah, 2020-04-23:
You should not have the "sdcard-xxx" folder in there, the contents of it must be at the root.

Joshenriquei, 2020-04-23:
[image]
Like this?

3djc, 2020-04-23:
No, here you should see RADIO MODELS SOUNDS IMAGES SCRIPTS...

Joshenriquei, 2020-04-23:
Sorry, I don't know how to do this. Could you help me?
[image]
For instance, like this?

3djc, 2020-04-23:
You need to move all this up in the root directory

Joshenriquei, 2020-04-23:
Solved! many,many thanks.

Joshenriquei, 2020-04-23:
[image]
Now I have programming Menu. Thanks