opentx/opentx #7548: external multi-modul: Protocol settings garbled

twistedwings, 2020-04-24:
Protocol settings of the external multi-modul are garbled after reading into companion.

Settings in X7 (protocol 62, XK):
[image]

After reading models into companion:
[image]

Taranis X7
iRangeX IRX4 FW v1.3.0.76
OpenTX 2.3.7
Companion 2.3.7

3djc, 2020-04-24:
Perfectly normal to not display newer protocols in companion. We probably need to alter number displayed (59 is what opentx uses to match multi 62), but we probably need to restrict the display to just protocol and subtype for unknow protocols

twistedwings, 2020-04-24:
I understand that unknown protocols are displayed by number instead of text.
I do not understand why is protocol 62 is changed to 59.
If I write the model back to the Tx, the protocol changes from 62 to 59 in the Tx also.

[image]

3djc, 2020-04-24:
Thats an issue. if you look on the tx , all frsky protocol are presented as one, you select frsky, and you have a choice of d8, x , v all in one. This ugly hack (like both Pascal and myself call it) create a progressive negative slide in protocol number (3 for protocols after 25, two from 15,..) Really ugly, but historical and cannot be changed before 2.4. Still I'll work on fixing the issue you esxperience comming from this

twistedwings, 2020-04-24:
I understand, that is really ugly.
Shall I close the issue or leave it open ?

3djc, 2020-04-24:
No no, I need to fix the fact it gets changed if you read and write

3djc, 2020-05-25:
Fixed at the same time as #7575 I believe

closed 2020-05-30 by 3djc