opentx/opentx #7549: macOS 10.15.4  Compiling OpenTx Error info = fatal error: 'Availability.h' file not found

1371030, 2020-04-24:
macOS:10.15.4
Xcode:11.4.1

cd build
cmake -DPCB=X10 -DPCBREV=T16 -DHELI=YES -DDEBUG=NO -DPYTHON_EXECUTABLE=/usr/local/cellar/python/3.7.7/bin/python3 -DCMAKE_BUILD_TYPE=Release ..

-- OpenTX 2.3.7 @8b080683
-- Python found, version: 3.7.3
-- Qt Version: 5.14.1
-- Foxlib found at /usr/local/lib/libFOX-1.6.dylib
-- Found SDL: /usr/local/lib/libSDL.dylib;-framework Cocoa (found version "1.2.15") 
-- SDL Lib: /usr/local/lib/libSDL.dylib;-framework Cocoa; Headers: /usr/local/include/SDL; Version: 1.2.15

Compiling Error info:
/usr/local/Cellar/gcc-arm-none-eabi-49/20150925/bin/../lib/gcc/arm-none-eabi/4.9.3/include/stddef.h:425:3: note: previous definition is here
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iterator:427:10: fatal error: 'Availability.h' file not found

kilrah, 2020-04-24:
https://stackoverflow.com/questions/58278260/cant-compile-a-c-program-on-a-mac-after-upgrading-to-catalina-10-15

1371030, 2020-04-26:
@kilrah Thanks,

But,

macOS 10.14.6
Xcode 11.0

make firmware
[  0%] Built target lua_export_t16
[  3%] Generating datacopy.cpp
/usr/local/Cellar/gcc-arm-none-eabi-49/20150925/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/include/sys/cdefs.h:371:9: warning: '__nonnull' macro redefined [-Wmacro-redefined]
  datastructs.h:25:10: note: in file included from datastructs.h:25:
  targets/horus/board.h:26:10: note: in file included from targets/horus/board.h:26:
  targets/common/arm/stm32/board_common.h:103:10: note: in file included from targets/common/arm/stm32/board_common.h:103:
  targets/horus/../simu/simpgmspace.h:32:10: note: in file included from targets/horus/../simu/simpgmspace.h:32:
  /usr/local/Cellar/gcc-arm-none-eabi-49/20150925/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/include/unistd.h:4:11: note: in file included from /usr/local/Cellar/gcc-arm-none-eabi-49/20150925/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/include/unistd.h:4:
  /usr/local/Cellar/gcc-arm-none-eabi-49/20150925/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/include/sys/unistd.h:11:10: note: in file included from /usr/local/Cellar/gcc-arm-none-eabi-49/20150925/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/include/sys/unistd.h:11:
  note: previous definition is here
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/11.0.0/include/__stddef_max_align_t.h:32:21: error: typedef redefinition with different types ('long double' vs 'struct max_align_t')
  datastructs.h:531:10: note: in file included from datastructs.h:531:
  ./gui/480x272/layout.h:24:10: note: in file included from ./gui/480x272/layout.h:24:
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/list:183:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/list:183:
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:653:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:653:
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/type_traits:406:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/type_traits:406:
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstddef:45:15: note: in file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstddef:45:
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/11.0.0/include/stddef.h:118:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/11.0.0/include/stddef.h:118:
  /usr/local/Cellar/gcc-arm-none-eabi-49/20150925/bin/../lib/gcc/arm-none-eabi/4.9.3/include/stddef.h:425:3: note: previous definition is here
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iterator:423:10: fatal error: 'Availability.h' file not found
  datastructs.h:531:10: note: in file included from datastructs.h:531:
  ./gui/480x272/layout.h:24:10: note: in file included from ./gui/480x272/layout.h:24:
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/list:183:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/list:183:
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:660:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:660:
make[3]: *** [radio/src/datacopy.cpp] Error 255
make[3]: *** Deleting file `radio/src/datacopy.cpp'
make[2]: *** [radio/src/CMakeFiles/datacopy.dir/all] Error 2
make[1]: *** [radio/src/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2

1371030, 2020-04-26:
generate_datacopy.py run error

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/11.0.0/include/__stddef_max_align_t.h:32:21: error: typedef redefinition with different types ('long double' vs 'struct max_align_t')

closed 2021-01-23 by stale[bot]