opentx/opentx #7550: DIY multimodule telemetry not work on 9XR pro

blairfancy, 2020-04-24:
I installed Latest openTX and latest DIY-multimodule FW and noticed that telemetry does not work.
tested with Hubsan and FRSKY protocole. Message displayed is No Telemetry in module configuration. Also noticed in telemetry menu that mode cannot be changed and stay at default

**To Reproduce**
Steps to reproduce the behavior:
1. Install multimodule on bay 
2. bind TX and RX
3. in telemetry find sensor
4. nothing appear

**Expected behavior**
RSSI feedback from module

**General information**
 - OpenTX version 2.3.7
 - Hardware 9X Pro multimodule TMX5 with FW 1.3.0.88

kilrah, 2020-04-24:
Do you have the right MP version (inverted/not inverted)? No idea which you'd need for a 9XR-Pro, so better try both.

blairfancy, 2020-04-25:
Tried both but none are working. Do you know on witch pin of ucontroler (cortex) is connected SPORT .
I m not able to see it in code of OpenTX

kilrah, 2020-04-25:
Seems it's USART2, PA5/PA6.

blairfancy, 2020-05-09:
Confirmed signal on PA5 (pin 53) but dows not handle it properly. I confirm that issue is related to openTX not hardware.
Module tested on Taranis and work properly (included telemetry)

closed 2021-02-04 by stale[bot]