opentx/opentx #7558: "Play Value" is not working in Companion

Waldo-P-Barnstormer, 2020-04-27:
OpenTx Companion 2.3.7 on Linux Mint 19.3 Cinnamon
X9D set as transmitter.
Special Function "Play Value" is not working. 
To reproduce go to Special Functions tab
1. Set "ON" as switch
2. Set "Play Value" as action. 
3. Set switch "SA" as Parameter.
4. Under Enable set repeat at 3s.
 
Run simulator and you should hear -100, 0, or 100 in the different SA switch positions. "Play Track" function works fine, only "Play Value" has issue.
My laptop dual boots into Win 7 or Linux and "Play Value" in the windows version of Companion 2.3.7 works fine. Trying to switch over to using Linux.

kilrah, 2020-04-28:
Missing the "SYSTEM" sound files in the SD folder?

Waldo-P-Barnstormer, 2020-04-28:
System files are there (screenshot provided). Sames ones I've used for many years, and work fine in Win 7 Companion 2.3.7. It is very odd, I checked again and it seems the system folder is not being seen at all. Countdown timers, altitude, voltage readouts, not working in the simulator. I tried copying the folder again from a flash drive backup, same result. What's odd is that all the files in the "en" folder play fine, just the subfolder system files not seen.
Wondering if this is some weird Linux issue, still getting used to it.

[image]

3djc, 2020-04-28:
Launch companion from command line, you’ll get more details

Waldo-P-Barnstormer, 2020-04-28:
OK, that helped solve the issue. Saw this in Terminal after loading a model
findTrueFileName(/home/jim/Taranis/(CLONE) FRSD1-4GB/SOUNDS/en/SYSTEM/0100.wav)
	not found
Changed my system folder name to all caps "SYSTEM" and it all works.
Had copied it over from my Windows install where it's in small letters, but I guess Linux needs it all caps.

3djc, 2020-04-29:
Linux has a case sensitive file system

closed 2020-04-29 by 3djc