opentx/opentx #7561: OpenTX defaults to DFU mode when powered off

Juzzle1, 2020-04-30:
There's likely an increase in the number of OpenTX users who will have their devices connected to a USB port while powered off, leaving them in DFU mode. Specifically devices that can charge via USB (Jumper T16 and Radiomaster T16), and possibly users who use their controllers as a HID device.

When flashing a flight controller with Betaflight Configurator (and likely other forks of Cleanflight), the Configurator application signals a DFU mode reboot via the user selected com port, and then finds and flashes the first DFU device it finds. If an OpenTX device is also plugged in but powered off, it is the OpenTX device that will be found first and written to.

I requested the Betaflight developers modify their firmware update routine to check and warn for existing DFU devices before putting the flight controller into DFU mode, however they were adamant this is an OpenTX problem as devices should never be in DFU mode unless it is specifically requested by the user.

I'm not sure what the answer is, or if you even think this is a problem, but posting here as I've done it myself, and think it could catch out a fair few others.

kilrah, 2020-04-30:
Well it is requested by the user by plugging the radio with power off...
Anyway it's done in hardware outside of our control so there's nothing we can change to it.

closed 2020-04-30 by kilrah

mpjf01, 2020-05-01:
Surely the fixes are simple enough. Either don't leave the transmitter plugged in to USB when messing about with flight controllers or don't charge it with the USB port on a computer in the first place, use one of the other options that everybody has, like a phone charger.  

The ability to flash OpenTx with the Tx powered off is important.

Juzzle1, 2020-05-01:
@kilrah  

_'Well it is requested by the user by plugging the radio with power off...'_

That really isn't obvious to anyone except developers.

@mpjf01

_'don't charge it with the USB port on a computer in the first place'_

Great idea - maybe you should tell some people that.

_The ability to flash OpenTx with the Tx powered off is important._

To who? Developers? Most people can update their radios by entering the OpenTX bootloader.

I've pointed out a legitimate problem here, take it or leave it.

3djc, 2020-05-01:
What part of "Anyway it's done in hardware outside of our control so there's nothing we can change to it" isn't clear ?

kilrah, 2020-05-01:
Yup, if you really want to change it you can but you'll have to hardware mod the radio, nothing we can do about it at our level.