opentx/opentx #7563: Logical switch always True if variable (V1) as Timer1  (V2) as  (any setting time)

njozsef, 2020-04-30:
**Describe the bug**
Timer1 countdown not triggering available: (9xrpro)

#     Function   V1            V2
L01  a<x           Timer1     0:04:10 [h:mm:ss]

**To Reproduce**
Steps to reproduce the behavior:
1. setting on remote ctrl (9xrpro)
2. set model menu countdown as Timer1 04:00
3. set L01 a<x Timer1 00:00:10 (10 seconds before end)
4. L01 always on any time.

**Screenshots**
-

**General information**
 - OpenTX version (2.3.7)
 - Hardware (9Xr Pro)
 - OpenTX Companion (e.g. 2.3.7 on W10)

Thank you.

3djc, 2020-05-01:
Maybe a 9xr pro issue, because it works fine here on several other radios

3djc, 2020-05-01:
Set at 0:00:10 as mentionned 

[image]

[image]

PR #7639 (2020-05-30): Fix 9x rotary encoder source 
This PR fixes #7563, #7564, #7433, #7199

The rotary encoder was removed from the model sources list in 8aa596e. The entry in the multi string and in companion sources needs to be removed as it was causing a string shift

I've added #7564 to 2.3.8 milestone. When affecting several issues, should I add the other issues to the milestone? or the PR instead?
closed 2020-05-30 by bsongis