opentx/opentx #7566: [Companion] Export LogViewer Graph as image

OpenMik, 2020-05-01:
Hi,
sometimes could be useful to export the visualized graph in companion logviewer as image (png, jpeg, bmp....)
for example using right mouse button menu "Export graph as image".
Now the option is to export the csv file and create a graph in Libreoffice Calc or make a screenshot and coutout it from saved image 
Thank you

3djc, 2020-05-01:
Most OS do allow to capture part of the screen in one way or another

OpenMik, 2020-05-01:
I am aware that many OS offer this option, but integrating this function could save some steps and improve the usability of LogViewer.
I understand that this option can be obtained in other ways and therefore may not be necessary.
Thanks anyway for the reply

closed 2021-01-30 by stale[bot]

OpenMik, 2021-01-30:
I still think that is  a useful option
but if no action was made, for developers is not a priority 
thanks anyway
OpenTx is a great job