opentx/opentx #7575: "Write Models and Settings to Radio" dialog corrupts model data

sbuhre, 2020-05-03:
The "Write Models and Settings to Radio" dialog corrupts "Multi Radio Protocol" of external Radio Module "DIY Multiprotocol Module"

Testcase:

Companion 2.3.7 on Windows 10 Build 1909 english
Open Companion
Create New Document (File|New)
Add Model (Wizard): Next, Next... Agree, Finish
Edit Model
Change "external Radio Module" to: "DIY Multiprotocol Module"
Change "Multi Radio Protocol" to "HoTT" (value is important, internal ID has to be > 16)
Close edit dialog
Open "Write Models and Settings to Radio" dialog
Close "Write Models and Settings to Radio" dialog
Edit Model
"Multi Radio Protocol" is corrupted: Changed from "HoTT" to "Devo"

3djc, 2020-05-03:
Please check n800, should be fixed already (and this is a duplicate of #7548

3djc, 2020-05-03:
Need to check how it behaves on companion model creation

3djc, 2020-05-03:
I'm sorry but I just tested N800, creating a model witrh wizard, selecting multi, then HoTT.

Transfer to radio (tested T16 intenral and X10 external), both ok on HoTT

sbuhre, 2020-05-03:
where do I get n800? I can only see n200.

https://downloads.open-tx.org/2.3/nightlies/companion/windows/

The problem is not the transfer to the TX. The settings are corrupted within companion. 
Just open the "write models and settings to radio" dialog. Don't do a transfer. Close it and edit the model again-> model corrupted.

3djc, 2020-05-03:
ok seem to apply only to non colorlcd radio, works fine on horus

sbuhre, 2020-05-03:
yes, tested with X-Lite Pro

3djc, 2020-05-04:
Fixed by https://github.com/opentx/opentx/commit/6f90454bd9dc58c5fff27ba3d167ce36997219bc

closed 2020-05-04 by 3djc