opentx/opentx #7579: Add a menu entry to create an empty model without the wizard

dgatf, 2020-05-04:
Hi,

Wouldn't be better to have the option to choose whether the wizard is used or not when creating a new model in opentx and companion?

The new entry could be 'Add empty model' in companion and 'Create empty model' in opentx

I know it is feasible to exit from the wizard but that's kind of a workaround

Sorry if this has been discussed before, couldn't find any related issue

Cheers,
Daniel

kilrah, 2020-05-04:
On companion you can already configure that. On the radio we considered that if you decided you didn't need the wizard you'd just delete it from the SD card.

dgatf, 2020-05-04:
Damn, I missed that option (...I shouldn't after the update of companion translation :flushed:). I know that if removed from the sd card it works without the wizard (though I didn't know it was on purpose), but still you may want to use it sometimes and sometimes not. IMHO it works better as a menu choice rather than an option. Though it's ok anyway and as this has been already decided I'm closing the issue. Thanks

closed 2020-05-04 by dgatf

kilrah, 2020-05-04:
Instead of deleting you can also rename the folder, that way it doesn't autorun but you can still go run it from the SD card browser.

dgatf, 2020-05-04:
Yes, that's better. And then should be easy to execute it from a script in the tools folder without the need of browsing the sd card (for the lazy like me)

kilrah, 2020-05-04:
I mean as soon as you've understood how OpenTX works the wizard is useless...

lshems, 2020-05-04:
Why not add an 'empty model' to the lua wizard as option. Best of both worlds.

dgatf, 2020-05-05:
Well, that's the same as exiting from the wizard. I found a bit annoying to exit from the wizard each time I create a model, but still want to have it available

lshems, 2020-05-05:
It looks cleaner

dgatf, 2020-05-05:
Agree