opentx/opentx #7580: Wizard update

dgatf, 2020-05-04:
Hi,

I'm thinking in adding an heli setup for the wizard (opentx and companion). It seems that something has been done in the past but it is not completed [https://github.com/opentx/opentx/issues/1647](url). Not sure if something prevents from making it feasible

There are some other issues that IMHO could be improved in the wizard:

- Add basic images for 1bit color displays in opentx
- The result is different in opentx and companion
- In opentx it is missing last screen options (throttle cut,...)
- Naming is different in opentx and companion
- Naming is incomplete in companion. It is missing output naming
- Actual naming in opentx rather go to outputs
- Throttle cut in companion should be an special function rather than a mix (as recommended for safety reasons)
- Flaps/brakes in companion are multiplexed and in opentx is a switch input

Some of the previous may be more of personal preferences on models setup though

I could work on those issues if some or all of those changes are welcome

Cheers,
Daniel

3djc, 2020-05-04:
1 bit images are really hard to get right so they display nicely but sure. There is nothing technicaly preventing Heli wizard from been made, it is just that nobody has done one. I think you'll find as many recommendation for throttle cut as there are experienced users (and then some more). I personnaly use Overide function with an additionnal safety so that it wont release if throttle isn't at -100, but thats just a personnal choice really. I know some power users run customized wizard to their models are setup the way they like it

kilrah, 2020-05-04:
> Throttle cut in companion should be an special function rather than a mix (as recommended for safety reasons)

It's the opposite, generally accepted that doing in mixer is safer, especially since there is an option to remove the "override" special function

> There is nothing technicaly preventing Heli wizard from been made

Pretty sure some of the settings relevant to heli are not yet available through lua

dgatf, 2020-05-04:
> 1 bit images are really hard to get right so they display nicely but sure

I use images in my x7 with this script [https://github.com/dgatf/ScrMod](url)
Not the best but is something

> I personnaly use Overide function with an additionnal safety so that it wont release if throttle isn't at -100, but thats just a personnal choice really. 

Absolutely right... I was thinking in safety switches from other firmware, which achieves the same as you said. In opentx I guess this would be by adding a sticky logical switch to the mix or the override function... going to update my models now. I agree, is a personal choice

dgatf, 2020-05-04:
> It's the opposite, generally accepted that doing in mixer is safer, especially since there is an option to remove the "override" special function

Can you explain please

> Pretty sure some of the settings relevant to heli are not yet available through lua

I think in the past it was missing the setCurve function in lua but now is available. Flight modes are not available but they are not mandatory. I personally don't use flight modes for helis

3djc, 2020-05-04:
There is a firmware option to not allow override function, so wizard relying on this to be available are in essence dangerous

dgatf, 2020-05-04:
Got it!

But why a firmware option to not allow override function?

kilrah, 2020-05-04:
Becasue some people consider it's bad / unsafe and don't want to see it...

dgatf, 2020-05-04:
Damn

dgatf, 2020-05-04:
> Pretty sure some of the settings relevant to heli are not yet available through lua

So far I've found that from lua it is missing the info for the flybar models. Flybars won't be feasible from lua
Flight modes, names can't be changed. Not really an issue though

lshems, 2020-05-04:
If you want to do flybar you can do so without the Heli Page, even swashplate.

I do find flightmodes an essential.

dgatf, 2020-05-05:
> If you want to do flybar you can do so without the Heli Page, even swashplate.

That's right. I didn't realized. It is not the standard way though but I'll try it

You can think in a similar way for the flightmodes, you can achieve the same with mixes and switches. Anyway I believe mixes for flightmodes can be done with lua, the only issue is that you can't rename the FM

lshems, 2020-05-05:
You can't set the switch to change flightmodes.

An example for swashplate can be found on www.justfly.solutions and the mixer can be found at my blog on www.rcgroups.com

Eldenroot, 2020-05-05:
Would be nice to add a heli into wizard :)

Eldenroot, 2020-11-02:
Please do not close :(

closed 2021-09-04 by stale[bot]